microsoft/AutonomousDrivingCookbook #1: error appeared when running TrainModel.ipynb

Mikemraz, 2017-12-25:
when running the following code in TrainModel.ipynb an error arose:
[sample_batch_train_data, sample_batch_test_data] = next(train_generator)

the error is:

> ValueError                                Traceback (most recent call last)
<ipython-input-6-6ddb22c8a6a3> in <module>()
     24     plt.show()
     25 
---> 26 [sample_batch_train_data, sample_batch_test_data] = next(train_generator)
     27 for i in range(0, 3, 1):
     28     draw_image_with_label(sample_batch_train_data[0][i], sample_batch_test_data[i])
D:\tools\anaconda\envs\ND-P2-python3\lib\site-packages\keras\preprocessing\image.py in __next__(self, *args, **kwargs)
    735 
    736     def __next__(self, *args, **kwargs):
--> 737         return self.next(*args, **kwargs)
    738 
    739 
~\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in next(self)
    240         # so it can be done in parallel
    241 
--> 242         return self.__get_indexes(index_array)
    243 
    244     def __get_indexes(self, index_array):
~\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in __get_indexes(self, index_array)
    243 
    244     def __get_indexes(self, index_array):
--> 245         index_array = sorted(index_array)
    246         if self.x_prev_states is not None:
    247             batch_x_images = np.zeros(tuple([self.batch_size]+ list(self.x_images.shape)[1:]),
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

please note that I have not added deleted any code in Generator.py file.
I do not know why this error arose, it makes no sense that sorted() function made this error.

mitchellspryn, 2017-12-28:
Can you give me the version of the following packages that you are using (this can be obtained by running `pip freeze`), as well as the python version you are running? For reference, here are the versions I am using:

numpy (1.13.3)
pandas (0.20.3)
keras (2.1.2)
tensorflow-gpu (1.1.0)

@adshar has run into the same issue in the past, and solved it by upgrading Keras to the latest (`pip install --upgrade Keras`)

Mikemraz, 2017-12-28:
@mitchellspryn Thanks for responding! I have fixed the bug by muself. When I use debug mode in jupyter notebook, I found that the object  `index_array` mentioned in the error message is a tuple and with a array in it. As showed below:

[image]

So I add a line in generator.py file, to get the first element(the array) of `index_array` and resigned it to `index_array` again. The problem solved. 

But how did adshar make it work without doing what I did? I mean, he only updated the version of keras. I am quite confused.

mitchellspryn, 2017-12-28:
Our generator is an extension of the Keras ImageDataGenerator(). In older versions of Keras, next() returns a tuple, as you discovered. In the later versions, next() only returns the masking index (the first item in the tuple). So, in addition to your workaround, updating Keras will solve this problem. 

Thanks for confirming that it works! I'll close the issue for now.

closed 2017-12-28 by mitchellspryn