microsoft/AutonomousDrivingCookbook #5: confirmConnection

zyxcambridge, 2018-01-30:
Waiting for connection: 
---------------------------------------------------------------------------
TransportError                            Traceback (most recent call last)
<ipython-input-2-fa91e588dfc3> in <module>()
      2 
      3 client = CarClient()
----> 4 client.confirmConnection()
      5 client.enableApiControl(True)
      6 car_controls = CarControls()

~/Desktop/AutonomousDrivingCookbook-master/AirSimE2EDeepLearning/AirSimClient.py in confirmConnection(self)
    158     def confirmConnection(self):
    159         print('Waiting for connection: ', end='')
--> 160         home = self.getHomeGeoPoint()
    161         while ((home.latitude == 0 and home.longitude == 0 and home.altitude == 0) or
    162                 math.isnan(home.latitude) or  math.isnan(home.longitude) or  math.isnan(home.altitude)):

~/Desktop/AutonomousDrivingCookbook-master/AirSimE2EDeepLearning/AirSimClient.py in getHomeGeoPoint(self)
    167 
    168     def getHomeGeoPoint(self):
--> 169         return GeoPoint.from_msgpack(self.client.call('getHomeGeoPoint'))
    170 
    171     # basic flight control

~/.conda/envs/tensorflow/lib/python3.5/site-packages/msgpackrpc/session.py in call(self, method, *args)
     39 
     40     def call(self, method, *args):
---> 41         return self.send_request(method, args).get()
     42 
     43     def call_async(self, method, *args):

~/.conda/envs/tensorflow/lib/python3.5/site-packages/msgpackrpc/future.py in get(self)
     41                 else:
     42                     if isinstance(self._error, error.RPCError):
---> 43                         raise self._error
     44                     else:
     45                         raise error.RPCError(self._error)

TransportError: Retry connection over the limit

mitchellspryn, 2018-01-30:
This means that our code cannot connect to AirSim. Did you start the executable?

demon221, 2018-02-01:
@mitchellspryn In the test ipynb which executable shall be started for connection to AirSim?

shrimalmadhur, 2018-02-01:
@demon221 You need to start the AirSim executable. The test code will connect to it. Follow the instructions [here](https://github.com/Microsoft/AirSim#how-to-get-it) to run that executable.

adshar, 2018-02-04:
@zyxcambridge As is mentioned in TestModel.ipynb, you have to start the simulator *before* you run the code.

@demon221 We have updated instructions in the readme for the standalone AirSim build to be used for this tutorial.

Closing this Issue now.

closed 2018-02-04 by adshar