microsoft/AutonomousDrivingCookbook #7: Right AirSim assets for this tutorial

vohtaski, 2018-02-02:
Which AirSim assets are you using?
https://github.com/Microsoft/AirSim/releases

I trained the model
```
Epoch 00024: val_loss improved from 0.00025 to 0.00024, saving model to model\models\model_model.24-0.0002403.h5
 - 57s - loss: 0.0017 - val_loss: 2.4029e-04
```
 and downloaded v1.1.7 LandscapeMountains.zip.

The code connects to it, and handles the car.
However, in the beginning of the simulation, there is the road in the forest (different
from the training pics), so immediately after several meters my car goes off the road
and is stuck next to a tree.

Am I doing something wrong?

adshar, 2018-02-03:
Thanks for bringing this to our attention. You are correct, this seems to be a versioning/asset issue. We are currently working on fixing this. Will follow up on this thread soon.

mitchellspryn, 2018-02-04:
It looks like there have been some changes between the official AirSim release and the version we used for development. In the official AirSim release, the car starts next to a nasty turn. Also, the map that we used was originally used in a hang gliding video game, so there are still some assets left there (e.g. the circles in the sky). The lighting also looks different.

[Here is the build we've been using](https://airsimtutorialdataset.blob.core.windows.net/e2edl/ADCookbookAirSim.7z). Download the zip and run "ADCookbook_Airsim_Mountain". We will update the instructions to fix this.

adshar, 2018-02-04:
The instructions have been fixed with information about the standalone build for this tutorial, thank you again for bringing this to our attention, closing the Issue now.

closed 2018-02-04 by adshar

vohtaski, 2018-02-05:
Great, thanks, it works!