microsoft/AutonomousDrivingCookbook #8: Run AirSim simulator on Mac

vshkola, 2018-02-19:
Hi,

I found this tutorial via AirSim repo. Fortunately, I was able to run AirSim on Mac, but this cookbook doesn't have any binaries except for windows, does it possible to include UnrealEngine sources to be able to build on my own? 

Best regards,
Vladyslav Shkola

themmer, 2018-03-02:
@vshkola  I'm guessing you would have to find a way around the powershell scripts (.ps1 file). I was wondering the same thing, but for linux instead of Mac.

mitchellspryn, 2018-03-03:
Porting the .ps1 script won't help - the binaries are for windows. 

The problem is that there is no binary version of RpcLib for linux distributions that we can just link with Unreal. We build the binaries from the windows version of the unreal editor. I've managed to cross-compile our binaries for Linux using some hacks, but that solution only works for a specific version of linux (i.e. I'd first build RpcLib on the target version of linux, copy the .so file to windows and add the proper linker flags. This wouldn't work in general). 

To properly fix this, we'd need to find a way to include RpcLib in the build process. This is a work in progress.

Parnia, 2018-09-26:
@mitchellspryn I also want to use this cookbook in Linux. It would be awesome if you could tell us the hack