microsoft/AutonomousDrivingCookbook #10: Version of packages

edimoon777, 2018-03-03:
I've tried to run 'TrainModel.ipynb' using Jupyter Notebook.
but It is difficult to install OpenCV libraries because of version matchings.

Can you tell me versions of additional packages such as openCV, CUDA, CuDNN?

I'm trying to run the code with
Python 3.5.2
Anaconda 4.1.1 (64bit)
CNTK 2.4
Keras 2.1.4
Tensorflow-gpu 1.5.0
CUDA9.1
CuDNN9.0
OpenCV??

Error message because of openCV as below.
-------------------------------------------------
C:\Anaconda3\lib\site-packages\cv2\__init__.py in <module>()
      2 import os
      3 
----> 4 from .cv2 import *
      5 from .data import *
      6 

ImportError: DLL load failed:
-------------------------------------------------

mitchellspryn, 2018-03-03:
I've seen this problem with OpenCV on windows before. Try one of [these wheels](https://www.lfd.uci.edu/~gohlke/pythonlibs/#opencv)?

anonymouslucky, 2018-03-03:
I am not getting Cooking package

mitchellspryn, 2018-03-04:
Can you post the error you are getting? AFAIK there is no "cooking" package.

anonymouslucky, 2018-03-05:
I got it sir, thanks for asking though. Actually it is a python script in
the repo and I had not realised that.

edimoon777, 2018-03-05:
I solved this issue.
But I don’t know exact solution because I tried many things from installation of python, tensorflow to anaconda.

I uninstalled pip and conda and then install packages only in conda environment.
After that installation of OpenCV 3.4.1 was successful. 

But I have to solve next problems with CUDA and CuDNN versions.
Information of detail versions of packages will be very helpful for beginners.

edimoon777, 2018-03-05:
I solved this issue.
My packages versions are listed as below (Windows 10 64bit).

Conda 4.4.11 (64bit)
Python 3.5.2
CNTK 2.4
Keras 2.1.4
Tensorflow-gpu 1.6.0
CUDA 9.1
CuDNN9.1
OpenCV 3.3.1
Opencv-python 3.4.1
opencv3 3.1.0
hdf5 1.10.1
h5py 2.7.1

anonymouslucky, 2018-03-05:
I got it where I was wrong.
The Cooking package is in the repo itself which I had not seen.

mitchellspryn, 2018-03-06:
Cool. Unfortunately, having an official image with these packages is a bit tricky, as for our other tutorial (the distributed RL one), CuDNN 9.1 is not installed by default on azure VMs (only 8, meaning that we need to use Tensorflow-gpu 1.4.x). 

Anyway, It sounds like all of the issues are solved, so I'm going to close this issue.

closed 2018-03-06 by mitchellspryn