microsoft/AutonomousDrivingCookbook #13: conversion to bad driving

rafi-vivanti, 2018-03-12:
Hi. Great work! I'm really enjoying your simulator and cookbook.

In the train phase, although there are 500 epochs, mine converges after 27-39 epochs due to "no improvement".  
Then in the test phase, the car do not keep track with the path and crashes very fast. Much less than the 5 minutes it should drive. 
I did not change the code, except for the TQDM  NotebookCallback I disabled. 
Any ideas about what's going on?

Thanks!

adshar, 2018-03-12:
Thanks for the kind words! 

How many times did this happen to you? Every time you kick off training weights are initialized differently, so maybe try running training again? Also, are you using the correct environment (landscape)? Lastly, you can change the earlyStopping callback to let the training go longer. 

Please know that 500 is just an upper limit and you will likely converge much earlier. Also, you probably did already, but please make sure you run all the preprocessing steps before you kick off training.

rafi-vivanti, 2018-03-13:
Hi. Thanks for the prompt reply!

I tried 6 times. I'm using mountain-landscape environment. It is different than the gif you present here. It starts with a right turn which my car over-turn it and crashes to the trees on the right. Also, I do not see the bridges you wrote about. 

I disabled the TQDMNotebookCallback since it crashes on:
"traitlets.traitlets.TraitError: The 'kernel' trait of a Comm instance must be a kernel... 'NoneType' was specified. Is it important for convergence?

adshar, 2018-03-13:
Oh I see what the problem is. You are using the wrong AirSim build. Looks like you downloaded it from the main AurSim repo. Please use the AirSim build we created for the cookbook. The download instructions are provided in the readme.

closed 2018-03-15 by adshar