microsoft/AutonomousDrivingCookbook #14: the generator for the behavior cloning

Fengmoon93, 2018-03-13:
/AutonomousDrivingCookbook-master/AirSimE2EDeepLearning/Generator.py", line 246, in __get_indexes
index_array = sorted(index_array)
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
I'm using Python 2.7.6
here is the index_array
(array([20087, 11112, 31001,  5888, 24707, 24742, 17835, 15747, 28776,
       19486,  7640, 28383, 20912,  8023,  8257, 13928,  6409, 22015,
       30856, 11651, 24858,  3563,   329, 21344,  2916, 12922, 24723,
        9054, 12824, 16085,  3248, 12290]), 0, 32)
where did you define the index_array?
I can't find it

adshar, 2018-03-13:
#1 

Please check the Issue linked above. You need to update Keras. Also, as noted in the instructions, you should be using Python 3.5 or above for the tutorials.

Fengmoon93, 2018-03-13:
thank you for the quick response~
I will try it.

closed 2018-03-13 by Fengmoon93