microsoft/AutonomousDrivingCookbook #16: missing parameters in LaunchLocalTrainingJob.ipynb

rafi-vivanti, 2018-03-19:
Great tutorial guys!

In the last section of LaunchLocalTrainingJob.ipynb , where you build the train.bat file, I think you're missing some parameters:
-local_run
-experiment_name

Also, if we run locally, why we need a batch file? can't we simply call app/distributed_agent directly?

Thanks!

mitchellspryn, 2018-03-19:
Funny you should mention this - we noticed this a few days ago. If you pull the branch mspryn_usability_fixes, it should be fixed. This has yet to be formally reviewed and merged to master, but I tried it on my box a few day ago, and it worked. 

If you look at the batch file, it is directly calling distributed_agent.py. The problem is that there are a bunch of command line parameters, and getting them right is a bit tricky. So we created the batch file to automate launching the job. Make sure to run the batch file from your activated anaconda environment.

mitchellspryn, 2018-03-19:
Looks like it's been merged. I'll close this issue.

closed 2018-03-19 by mitchellspryn