microsoft/AutonomousDrivingCookbook #18: Error occurs when running TrainModel

BrianN92, 2018-03-20:
## Problem description
Hi all, I tried to run **"TrainModel"** today. Everything went well until an error occurred showed that **"axis 3 is out of bounds for array of dimension 3"** when running to **"[sample_batch_train_data, sample_batch_test_data] = next(train_generator)"**

## Problem details
I didn't make any change to the original file. It looks like that the issue is about next function in "image.py" file. 
[error](https://user-images.githubusercontent.com/21299740/37629902-cb9d2b44-2bb7-11e8-9099-93bb317ffed6.PNG)

## Exepriment/Environment details
* Tutorial used: *AirSimE2EDeepLearning*
* Versions of artifacts used (if applicable): *Python 3.6, Keras 2.1.5*

mitchellspryn, 2018-03-20:
Please try [keras 2.1.2](https://github.com/Microsoft/AutonomousDrivingCookbook/issues/1), and see if the error still happens.

BrianN92, 2018-03-20:
@mitchellspryn 
That works! Thank you! Although I have no idea why the most recent version does not work.

closed 2018-03-20 by BrianN92

mitchellspryn, 2018-04-01:
[Recycling old issues](https://github.com/Microsoft/AutonomousDrivingCookbook/issues/1).