microsoft/AutonomousDrivingCookbook #20: random weights works well,  Transfer Learning failed

danfouer, 2018-03-23:
## Problem description
I used DistributedRL Tutorial with random weights for two days, then I stop it and began to use Transfer learning weights, however, it seems cudnn is unreachable. I have cudnn5.1 and 6.0 for cund8.0, and cudnn7 for cudn 9.1

## Problem details
Loading weights from my_model_weights.h5...
Current working dir is D:\AIProject\AutonomousDrivingCookbook\DistributedRL\Share\scripts_downpour\app
First layer:
[[[[ 0.03894855 -0.02513464  0.15417658 -0.0026583  -0.07323869  0.12281457
    -0.07029789  0.15666577 -0.12889594  0.11235218 -0.12835188 -0.15529177
    -0.12245906  0.15574838 -0.16965055 -0.00851028]
   [-0.13678242  0.17681859  0.17137034  0.05008856  0.11469441  0.06297248
     0.12309571  0.09143925 -0.16798416  0.16038734  0.1695745   0.13292642
    -0.02244652 -0.04735599 -0.00631354  0.03638385]
   [ 0.19165193 -0.14834613  0.17736149  0.18141983  0.04048967 -0.20928982
    -0.11057959  0.06388518 -0.17496441  0.17916133 -0.01306503 -0.07610687
     0.1388101  -0.08880593 -0.03391987  0.15310904]]

  [[-0.03631234  0.04950815 -0.10022049 -0.11774249  0.15822944  0.06053431
    -0.04932768  0.1228191  -0.05324809 -0.09266006 -0.05179078 -0.11411756
    -0.095482    0.11878865 -0.08317431  0.13736732]
   [-0.17175385  0.04252055  0.15489234  0.03725579  0.01006478 -0.15945505
    -0.02473786 -0.15020007 -0.09039721 -0.16555463  0.13943027  0.0628512
     0.15334439  0.11108618 -0.00479105 -0.10645811]
   [ 0.07858201  0.03983586 -0.10747178  0.17793405  0.05197094  0.01532679
     0.05628907 -0.01228614  0.0297354  -0.11679398 -0.15627411 -0.1907029
     0.13904789 -0.0640917   0.02927678 -0.11179238]]

  [[-0.14073054  0.12452096 -0.05107016 -0.06377813 -0.05155271  0.13970849
    -0.18506047  0.02905693  0.05233993 -0.06130159 -0.07949064 -0.11921628
     0.13336256  0.18380319 -0.14173897 -0.12382756]
   [-0.09227199 -0.08912648  0.04024128 -0.16723497  0.03934036 -0.17860699
    -0.12220516 -0.10621318  0.05502975 -0.09897196 -0.04708755 -0.01756219
    -0.04116414 -0.11901203  0.13823269  0.15401933]
   [ 0.19709682 -0.19583263  0.07211456 -0.09657288  0.12548693 -0.17089447
     0.09015843  0.10009236  0.09117922 -0.05886658  0.19111662 -0.15089129
     0.09723601  0.06179318  0.04969154 -0.02905231]]]

 [[[-0.15161256  0.18108158 -0.08198252  0.03675875  0.08679089 -0.01822327
     0.06029394 -0.01858982  0.10493858 -0.13273385 -0.01893125 -0.0892955
     0.15530367 -0.12394772 -0.01735278 -0.09993808]
   [-0.13090514  0.00258087  0.15755838  0.08856946 -0.14106721 -0.10148135
     0.14753282 -0.0588728   0.07801725  0.18269575 -0.05899098 -0.13561122
    -0.03079795  0.18038721  0.12587661 -0.04889482]
   [ 0.1891939   0.16006617 -0.14898627  0.02304085 -0.11231866  0.1505771
     0.09320717  0.14030428 -0.07523882 -0.13361876 -0.15507735  0.17406569
     0.2038019   0.14741437  0.09891715 -0.00350525]]

  [[-0.06804357  0.02128448  0.00749603 -0.01405669  0.08071361 -0.13800594
     0.04556615  0.14130948 -0.02284213  0.15676719 -0.07210111  0.07653186
    -0.08501238  0.05455744  0.02490745 -0.15323325]
   [-0.15171303 -0.15090781  0.02632188  0.1974992  -0.16909063  0.06349774
     0.00563027  0.00426664 -0.11086674  0.1002009   0.14993776  0.09362567
    -0.03243918 -0.1058441  -0.17391473 -0.15979324]
   [ 0.09868525  0.13697761  0.10584432  0.132819   -0.0173201   0.08517482
    -0.09365354  0.03072184 -0.14881726 -0.01704379  0.0856963   0.01867709
    -0.04958879 -0.02375238  0.08077671 -0.17336646]]

  [[ 0.00752353  0.13349725 -0.16632664 -0.15007068  0.09292189  0.02270007
     0.08131488  0.15506698  0.15182415 -0.12181804  0.00567346 -0.12908186
    -0.02691863  0.15363601 -0.13808645  0.15727647]
   [ 0.14065173 -0.15125008 -0.05247576 -0.00246365 -0.12311495  0.16167094
    -0.02993548 -0.07256303  0.17449853  0.04090489  0.02223159 -0.05246314
    -0.06843664  0.15503356 -0.04727919 -0.17540433]
   [ 0.00120394  0.11387651  0.0409864   0.16292714  0.088181   -0.01079098
     0.15381627 -0.0891685   0.15814507 -0.12307656  0.13868488 -0.09270314
     0.00978417 -0.04605329  0.18279535 -0.13144042]]]

 [[[-0.15878747  0.03906631 -0.14726698 -0.13384612 -0.17665805 -0.15925416
    -0.00425105  0.02009274  0.09187473 -0.06346051  0.09910821  0.08797219
     0.14174503 -0.11942405  0.07518984  0.17416692]
   [ 0.05318895 -0.04195464  0.07309341 -0.08091445  0.08116957 -0.14057948
    -0.14835921 -0.07002999  0.11191972 -0.12638785 -0.08042201  0.18678014
    -0.1666165  -0.10577761  0.01899135 -0.05747586]
   [ 0.14029655  0.07926279 -0.15504895  0.16135754  0.14789698 -0.14238183
    -0.16055839  0.17460454  0.18023777 -0.0956251  -0.06245841  0.17527607
     0.12980177  0.02224305 -0.01771989 -0.11362264]]

  [[-0.01645875  0.17362995  0.04088786 -0.19339693 -0.13268843 -0.10124306
    -0.03237866 -0.02942094  0.15809548 -0.03399407  0.08054227  0.01921189
     0.02275016  0.15085441 -0.0945925   0.02212913]
   [ 0.13483758  0.0636002  -0.00913755  0.05353547  0.06470659 -0.05704467
     0.09975314  0.14212339 -0.13294634  0.04109536  0.08934512 -0.02663192
    -0.13352115 -0.02337635 -0.00685774  0.18906754]
   [-0.09998655  0.06588923 -0.01494381 -0.06959759 -0.0002833   0.05629796
     0.00485297  0.09865069 -0.11832282  0.00457611  0.05286699  0.15218307
     0.02567867 -0.01757005  0.18179703 -0.04168937]]

  [[ 0.15269685 -0.10070352 -0.16917658 -0.01993535 -0.11679676  0.09519561
     0.03552201  0.00036524  0.11301665  0.09974247 -0.19909017  0.16585428
     0.15420395 -0.09478516  0.03247167  0.11814672]
   [ 0.01920974  0.12406559 -0.0675544  -0.16710982  0.0430256  -0.03124766
     0.16766725  0.13785729 -0.07411029  0.127984    0.13921042 -0.06976632
    -0.14361116 -0.15726452 -0.05329617 -0.08604978]
   [-0.01553447 -0.18177256 -0.12387755  0.0188439  -0.08041562  0.20500283
    -0.1698696  -0.13808864 -0.16904303  0.1554161   0.05020312  0.0638795
    -0.08212703  0.00350113 -0.08972123  0.13617031]]]]
Attempting to connect to AirSim (attempt 0)
WARNING:tornado.general:Connect error on fd 2068: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2068: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2068: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2068: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2068: WSAECONNREFUSED
Waiting for connection: Failed to connect.
Waiting a few seconds.
Attempting to connect to AirSim (attempt 1)
Waiting for connection:
Connected!
Filling replay memory...
Running Airsim Epoch.
Running AirSim epoch.
Getting Pose
Waiting for momentum to die
Resetting
Running car for a few seconds...
Start time: 2018-03-23 22:28:00.290143, end time: 2018-03-23 22:28:01.669712
Percent random actions: 1.0
Num total actions: 24
Replay memory now contains 24 members. (48.0% full)
Running Airsim Epoch.
Running AirSim epoch.
Getting Pose
Waiting for momentum to die
Resetting
Running car for a few seconds...
Start time: 2018-03-23 22:28:07.721129, end time: 2018-03-23 22:28:09.958872
Percent random actions: 1.0
Num total actions: 49
Replay memory now contains 50 members. (100.0% full)
Replay memory filled. Starting main loop...
Running Airsim Epoch.
Running AirSim epoch.
Getting Pose
Waiting for momentum to die
Resetting
Running car for a few seconds...
Start time: 2018-03-23 22:28:16.003218, end time: 2018-03-23 22:28:19.942759
Percent random actions: 1.0
Num total actions: 77
Generating 77 minibatches...
Sampling Experiences.
Publishing AirSim Epoch.
Publishing epoch data and getting latest model from parameter server...
START GET GRADIENT UPDATE DEBUG
2018-03-24 06:28:26.109428: E c:\l\work\tensorflow-1.1.0\tensorflow\stream_executor\cuda\cuda_dnn.cc:359] could not create cudnn handle: CUDNN_STATUS_NOT_INITIALIZED
2018-03-24 06:28:26.114238: E c:\l\work\tensorflow-1.1.0\tensorflow\stream_executor\cuda\cuda_dnn.cc:366] error retrieving driver version: Unimplemented: kernel reported driver version not implemented on Windows
2018-03-24 06:28:26.122436: E c:\l\work\tensorflow-1.1.0\tensorflow\stream_executor\cuda\cuda_dnn.cc:326] could not destroy cudnn handle: CUDNN_STATUS_BAD_PARAM
2018-03-24 06:28:26.130426: F c:\l\work\tensorflow-1.1.0\tensorflow\core\kernels\conv_ops.cc:659] Check failed: stream->parent()->GetConvolveAlgorithms(&algorithms)

## Exepriment/Environment details
* Tutorial used:  DistributedRL
* Environment used: local_run
* Versions of artifacts used (if applicable): Python 3.5, Keras 2.1.2

mitchellspryn, 2018-04-01:
@danfouer Did the fix in #17 solve the issue?

mitchellspryn, 2018-04-05:
This issue appears to be stale. I'll close it for now. @danfouer, if you are still having trouble, feel free to either reopen or file another issue if you're hitting another error.

closed 2018-04-05 by mitchellspryn