microsoft/AutonomousDrivingCookbook #25: error train the model

hyybuaa, 2018-03-31:
python35 cudnn5.1 keras2.1.2 numpy (1.14.2)
Traceback (most recent call last):
  File "C:\Users\oleg\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 526, in get
    value = obj._trait_values[self.name]
KeyError: 'layout'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\oleg\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 526, in get
    value = obj._trait_values[self.name]
KeyError: 'kernel'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "E:/AutonomousDriving/AirSimE2EDeepLearning/test2.py", line 122, in <module>
    validation_data=eval_generator, validation_steps=num_eval_examples//batch_size, verbose=2)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\keras\legacy\interfaces.py", line 87, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\keras\engine\training.py", line 2072, in fit_generator
    callbacks.on_train_begin()
  File "C:\Users\oleg\Anaconda3\lib\site-packages\keras\callbacks.py", line 126, in on_train_begin
    callback.on_train_begin(logs)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\keras_tqdm\tqdm_callback.py", line 129, in on_train_begin
    total=epochs)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\keras_tqdm\tqdm_callback.py", line 67, in build_tqdm_outer
    return self.tqdm(desc=desc, total=total, leave=self.leave_outer)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\keras_tqdm\tqdm_notebook_callback.py", line 33, in tqdm
    return tqdm_notebook(desc=desc, total=total, leave=leave)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\tqdm\__init__.py", line 22, in tqdm_notebook
    return _tqdm_notebook(*args, **kwargs)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\tqdm\_tqdm_notebook.py", line 176, in __init__
    self.sp = self.status_printer(self.fp, self.total, self.desc)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\tqdm\_tqdm_notebook.py", line 96, in status_printer
    pbar = IntProgress(min=0, max=total)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\ipywidgets\widgets\widget_int.py", line 57, in __init__
    super(cls, self).__init__(**kwargs)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\ipywidgets\widgets\widget_int.py", line 95, in __init__
    super(_BoundedInt, self).__init__(**kwargs)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\ipywidgets\widgets\widget_int.py", line 76, in __init__
    super(_Int, self).__init__(**kwargs)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\ipywidgets\widgets\domwidget.py", line 90, in __init__
    super(DOMWidget, self).__init__(*pargs, **kwargs)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\ipywidgets\widgets\widget.py", line 184, in __init__
    self.open()
  File "C:\Users\oleg\Anaconda3\lib\site-packages\ipywidgets\widgets\widget.py", line 197, in open
    state, buffer_keys, buffers = self._split_state_buffers(self.get_state())
  File "C:\Users\oleg\Anaconda3\lib\site-packages\ipywidgets\widgets\widget.py", line 291, in get_state
    value = to_json(getattr(self, k), self)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 554, in __get__
    return self.get(obj, cls)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 533, in get
    value = self._validate(obj, dynamic_default())
  File "C:\Users\oleg\Anaconda3\lib\site-packages\ipywidgets\widgets\domwidget.py", line 23, in _layout_default
    return Layout()
  File "C:\Users\oleg\Anaconda3\lib\site-packages\ipywidgets\widgets\widget.py", line 184, in __init__
    self.open()
  File "C:\Users\oleg\Anaconda3\lib\site-packages\ipywidgets\widgets\widget.py", line 203, in open
    self.comm = Comm(**args)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\ipykernel\comm\comm.py", line 56, in __init__
    self.open(data=data, metadata=metadata, buffers=buffers)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\ipykernel\comm\comm.py", line 83, in open
    comm_manager = getattr(self.kernel, 'comm_manager', None)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 554, in __get__
    return self.get(obj, cls)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 533, in get
    value = self._validate(obj, dynamic_default())
  File "C:\Users\oleg\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 589, in _validate
    value = self.validate(obj, value)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 1681, in validate
    self.error(obj, value)
  File "C:\Users\oleg\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 1528, in error
    raise TraitError(e)
traitlets.traitlets.TraitError: The 'kernel' trait of a Comm instance must be a Kernel, but a value of class 'NoneType' (i.e. None) was specified.

mitchellspryn, 2018-04-01:
This seems like something is wrong with your Keras install. Can you try reinstalling in a clean venv from the InstallPackages.py?

hyybuaa, 2018-04-01:
@mitchellspryn  sorry ,i'm a newer and don't know how to reinstall in a clean env

hyybuaa, 2018-04-01:
sorry,i don't use jupyter notebook.I edit it in pychram ,so error

mitchellspryn, 2018-04-01:
It sounds like your environment is not set up properly. Please follow the instructions that are specified in the "Prerequisites and Setup" section of the main README page [here](https://github.com/Microsoft/AutonomousDrivingCookbook/tree/master/AirSimE2EDeepLearning). This explains in detail how to install anaconda and the dependencies. 

Also, the tutorial files are jupyter notebooks. AFAIK, they won't work directly in pycharm unless you convert them to standard python files. Also, I don't know how pycharm interacts with anaconda virtual environments - you're probably better off just using the notebooks. Is there a specific reason you're trying to use pycharm?

mitchellspryn, 2018-04-05:
@hyybuaa, were you able to run the notebooks?

hyybuaa, 2018-04-05:
@mitchellspryn yes, I can

mitchellspryn, 2018-04-05:
Great. I'll close this issue, then. Thanks for confirming!

closed 2018-04-05 by mitchellspryn