microsoft/AutonomousDrivingCookbook #27: Steering angle values have not been completely recorded.

BrianN92, 2018-04-03:
I tried to run this Autonomous driving cookbook on other maps like the neighborhood. However, the steering angle values in " airsim_rec.txt" only have three values: -1, 0, and 1. I have attached one file below.
[airsim_rec.txt](https://github.com/Microsoft/AutonomousDrivingCookbook/files/1869896/airsim_rec.txt)

Don't know how to deal with it. Has anyone ever met this issue? Thank you！

[image]
[image]

mitchellspryn, 2018-04-05:
What are you using to control the car? If you're using the keyboard to drive, then there will only be -1, 0, 1 steering angles, as a key is only ever up or down. We used [this steering wheel](https://www.walmart.com/ip/Thrustmaster-Ferrari-Red-Legend-Race-Wheel-PS3-PC/30824036?wmlspartner=wlpa&selectedSellerId=0&adid=22222222227021749969&wl0=&wl1=g&wl2=c&wl3=40342616792&wl4=pla-78302436992&wl5=9033315&wl6=&wl7=&wl8=&wl9=pla&wl10=8175035&wl11=online&wl12=30824036&wl13=&veh=sem) when generating data.

BrianN92, 2018-04-05:
@mitchellspryn Thanks for the reminder! Yes I used keyboard to drive. I tried the Logitech G29 steering wheel before, but it could not use on Airsim. Which is another unsolved issue, haha.

mitchellspryn, 2018-04-06:
Are you using [x360ce](https://www.x360ce.com/) to map the inputs to xbox controller inputs? We had to do that.

mitchellspryn, 2018-04-07:
> @mitchellspryn <https://github.com/mitchellspryn> Yes, I tried x360ce to
> map the inputs, but it still could not work.

Did you put the x360ce exe in the same directory as the AirSim binary? I
think I recall having to do that.

BrianN92, 2018-04-08:
Yes, I did as you said. It did not work.

BrianN92, 2018-04-24:
@mitchellspryn I bought [the steering wheel](https://www.walmart.com/ip/Thrustmaster-Ferrari-Red-Legend-Race-Wheel-PS3-PC/30824036?wmlspartner=wlpa&selectedSellerId=0&adid=22222222227021749969&wl0=&wl1=g&wl2=c&wl3=40342616792&wl4=pla-78302436992&wl5=9033315&wl6=&wl7=&wl8=&wl9=pla&wl10=8175035&wl11=online&wl12=30824036&wl13=&veh=sem) you mentioned above. It did not work. Every time I connect the steering wheel with Airsim, the vehicle keep turning left with the "steering value: -1.25". The  "Accel" value was 0.5 all the time. I cannot use the wheel to control car's direction, but I can do it by pushing the braking panels. Besides, there was no values of "steering', "throttle" or "break" inside the "rec.txt" file. It was so weird, so could you tell me how did you use this steering wheels to control the car? 

I did put the x360ce exe in the same directory as the AirSim binary.

BrianN92, 2018-04-24:
I found this code in CarPawn.cpp. Seams like it is related to my problems

[image]

mitchellspryn, 2018-04-24:
Hm. Are you using the binary that we provided, or the one from the AirSim page? The version we've tested with is a few iterations behind, and doesn't have that code...

BrianN92, 2018-04-24:
Actually, I tried both. They had the same problems. I added "PhysXCar" : {
    "RC": {
        "RemoteControlID": 0
    }
} to "AD_Cookbook_Start_AirSim"file and put the x360ce exe in the "AD_Cookbook_AirSim" folder.

mitchellspryn, 2018-04-24:
Wait. What? Adding that json should either do nothing or cause a syntax error; it's a powershell script. 

You need to put the x360ce in the same directory as the binary that is running. I think that would be $/AD_Cookbook_AirSim/JustAssets/Binaries/Win64, not the root directory. Also, make sure that when you push buttons on the steering wheel, the corresponding buttons light up in the x360ce GUI.

BrianN92, 2018-04-24:
Fixed!! I deleted that json and put the x360ce in the correct directory. Now I can use the steering wheel to control the car! Thank you so much! I could have a good sleep now.

closed 2018-04-24 by BrianN92