microsoft/AutonomousDrivingCookbook #28: How long it take you to train a epoch, I take 10min,it means 80 hours for 500 epochs

hyybuaa, 2018-04-03:
it seems 45min for you 
my env: gipu=gtx1070 keras=2.1.2 tersonflow-gpu cudn=v8 cudnn=5.1

mitchellspryn, 2018-04-05:
Keep in mind you probably won't be training all 500 epochs, you'll likely converge after 30-40. This is why we included the [Early Stopping callback](https://keras.io/callbacks/#earlystopping).

hyybuaa, 2018-04-05:
@mitchellspryn  yes,I trained 25 epochs,and val_loss about 0.0002, but the result is bad.

mitchellspryn, 2018-04-05:
Can you describe a bit more what you mean by "bad"? When running the last cell of the notebook, do the labels and the predicted values line up?

Also, if you retrain the model, do you repeatedly hit this bad result, or did you only try training once? We don't have a very large dataset, so it's possible that you've converged into a local minimum. 

Also, dumb question, but are you using the binaries we provided or the downloads from the AirSim repo?

hyybuaa, 2018-04-05:
@mitchellspryn  About a minute after the test, the car crashed or pulled out of the road and stopped

hyybuaa, 2018-04-05:
@mitchellspryn  i use the dataset that downloads from the AirSim repo

mitchellspryn, 2018-04-05:
I was asking about the binary executable, not the dataset. If you downloaded the landscape.exe binary from the AirSim repo, please [download and use ours](https://airsimtutorialdataset.blob.core.windows.net/e2edl/AD_Cookbook_AirSim.7z), as the tutorial suggests. The reason they are different is that in the one on the AirSim repo, the car starts in a different place and the lighting is very different. There are other minor differences as well. 

The car probably won't make it the full length of the road with the stock model - the best runs I got were 5-6 min. There's always room for improvement :).

mitchellspryn, 2018-04-05:
Ok. Then it should eventually work. We only have 25k images in our dataset IIRC, so it may very well be that you've converged to a suboptimal minimum. I'd recommend trying it once or twice more, and if it still doesn't follow the road at all, then we can investigate further.

hyybuaa, 2018-04-05:
ok ,I will retrain it

closed 2018-04-12 by mitchellspryn