microsoft/AutonomousDrivingCookbook #30: Error in showing some training images in AirSimE2EDeepLearning repo

mmmmming18, 2018-04-05:
AxisError                                 Traceback (most recent call last)
<ipython-input-11-6ddb22c8a6a3> in <module>()
     24     plt.show()
     25 
---> 26 [sample_batch_train_data, sample_batch_test_data] = next(train_generator)
     27 for i in range(0, 3, 1):
     28     draw_image_with_label(sample_batch_train_data[0][i], sample_batch_test_data[i])

~\Anaconda3\lib\site-packages\keras\preprocessing\image.py in __next__(self, *args, **kwargs)
    858 
    859     def __next__(self, *args, **kwargs):
--> 860         return self.next(*args, **kwargs)
    861 
    862     def _get_batches_of_transformed_samples(self, index_array):

~\tmp\AutonomousDrivingCookbook-master\AirSimE2EDeepLearning\Generator.py in next(self)
    240         # so it can be done in parallel
    241 
--> 242         return self.__get_indexes(index_array)
    243 
    244     def __get_indexes(self, index_array):

~\tmp\AutonomousDrivingCookbook-master\AirSimE2EDeepLearning\Generator.py in __get_indexes(self, index_array)
    262                 x_images = x_images[self.roi[0]:self.roi[1], self.roi[2]:self.roi[3], :]
    263 
--> 264             transformed = self.image_data_generator.random_transform_with_states(x_images.astype(K.floatx()))
    265             x_images = transformed[0]
    266             is_horiz_flipped.append(transformed[1])

~\tmp\AutonomousDrivingCookbook-master\AirSimE2EDeepLearning\Generator.py in random_transform_with_states(self, x, seed)
    138             x = image.random_channel_shift(x,
    139                                      self.channel_shift_range,
--> 140                                      img_channel_axis)
    141         if self.horizontal_flip:
    142             if np.random.random() < 0.5:

~\Anaconda3\lib\site-packages\keras\preprocessing\image.py in random_channel_shift(x, intensity, channel_axis)
    177 
    178 def random_channel_shift(x, intensity, channel_axis=0):
--> 179     x = np.rollaxis(x, channel_axis, 0)
    180     min_x, max_x = np.min(x), np.max(x)
    181     channel_images = [np.clip(x_channel + np.random.uniform(-intensity, intensity), min_x, max_x)

~\Anaconda3\lib\site-packages\numpy\core\numeric.py in rollaxis(a, axis, start)
   1470     """
   1471     n = a.ndim
-> 1472     axis = normalize_axis_index(axis, n)
   1473     if start < 0:
   1474         start += n

AxisError: axis 3 is out of bounds for array of dimension 3

BrianN92, 2018-04-05:
Hi, I met this issue before. Please reinstall the Keras with 2.1.2 version. I fixed my problem by this way.

mitchellspryn, 2018-04-06:
Duplicate of [Issue 1](https://github.com/Microsoft/AutonomousDrivingCookbook/issues/1). Install keras 2.1.2 as @BrianN92  suggested.

closed 2018-04-06 by mitchellspryn