microsoft/AutonomousDrivingCookbook #32: For local DistributedRL, How to generate pretrain_model_weights.h5?

Ryuk03, 2018-04-12:
**Your issue may already be reported! Please make sure to search all open and closed issues before starting a new one.**

Please fill out the sections below so we can understand your issue better and resolve it quickly.

## Problem description
*(Please provide a 2-3 sentence description of your problem. Be concise to ensure this description is useful for future users who might run into the same issue.)*

1.Goal: I was trying to use RL in my own unreal environment to make the car following a trajectory(a set of positions) with the API controller in a constant speed.
   PS: If there is a better solution, please let me know :)
2. I use the trajectory as the reward_road to train the car to follow.
[image]

3. how should I generated the `pretrain_model_weights` for transfer training?
`weights_path = os.path.join(os.getcwd(), 'Share\\data\\pretrain_model_weights.h5')`

4. When I run the DistributedRL in neighborhood environment, I notices that the car driving on a curly path (that keep steering all the time). Is there a way that I can reduce the steering behavior when the car is driving?
## Problem details
*(Please describe your problem in as much detail as possible here. Make sure to include screenshots, code snippets, error messages, links and anything else you think will help us understand your problem better. If applicable, please also provide us a list of steps to reproduce your problem.)*

## Exepriment/Environment details
* Tutorial used: *(For example, AirSimE2EDeepLearning, DistributedRL etc.)*
DistributedRL 
* Environment used: *(For example, landscape, city, hawaii etc.)*
* Versions of artifacts used (if applicable): *(For example, Python 3.5, Keras 2.1.2 etc.)*

mitchellspryn, 2018-04-12:
I assume that you just have a single road that turns to the right in your own unreal environment? If your environment is that simple, you probably won't need to do pretraining, the model should learn well enough from the ground up. However, if you really want to try it, you'l need to drive that road a bunch of times and record your inputs (using the record button), and train the [end-to-end](https://github.com/Microsoft/AutonomousDrivingCookbook/tree/master/AirSimE2EDeepLearning) model on that dataset. 

Yes, the car will turn sharply at first. That's because it's making random decisions. As the model converges, it'll follow a more sane trajectory.

adshar, 2018-04-12:
Looks to me you just need better training. The network we used in the tutorials is pretty basic. Did you try playing around with it? Also, keep in mind the goal of transfer learning. You are trying to transfer learned representations to a new problem, so you don't have to necessarily do both supervised learning and RL in the same environment. You are extracting scene features for driving, so using the model we trained in the mountain environment should work well to do transfer learning for your RL algorithm in this new environment.

If you did want to do supervised learning in this environment, I would suggest a few things:
1. Get more data
2. You only have one 90 degree turn in this environment, everything else is a straight road. Remember the discussion in the End-to-end tutorial, your dataset might be imbalanced as it is not seeing the turn enough number of times. The swerve data helps the car stay come back to the road if it slides a little but you also need to account for that single sharp 90 degree turn. I would suggest collecting more data from that specific part of the road.
 
Additionally, I can see the model getting confused by the color of the road, color of the shadow and the color of the surroundings as they are all very similar. Have you done something to account for this?

mitchellspryn, 2018-04-21:
Closing due to staleness. 

@Ryuk03, please reopen if you need further assistance.

closed 2018-04-21 by mitchellspryn