microsoft/AutonomousDrivingCookbook #35: Update Airsim Environments

gusnitaa, 2018-04-23:
Hello,

I have a question about the environments used in this cookbook. Do you update this link https://airsimtutorialdataset.blob.core.windows.net/e2edl/AD_Cookbook_AirSim.7z when new release appears on the github Microsoft/Airsim (https://github.com/Microsoft/AirSim/releases) ? I've been working on this cookbook for a while and have notified that the road_points / reward_points in your NH environment are not the same as in NH from github Microsoft / Airsim. The physic of the car is also different.

mitchellspryn, 2018-04-24:
The point of providing the executables is to provide you with something that just works out of the box, and is independent of the changes made by AirSim. Updating to the latest version of AirSim is not as simple as just rebuilding the executables with the latest version of the plugin; we'd have to ensure that all of the tutorials work as well, which is very time consuming. You can certainly try using the latest version of the AirSim binaries that are made available on their GitHub page, but there will be differences that will affect the quality of your final results / models. You may need to regenerate datasets or retune models. 

Is there a particular feature that is missing that makes the tutorials unworkable in our binary releases?

gusnitaa, 2018-04-24:
Thanks for the answer. You are right about updating. 
It can change radically the quality of the final results.
The problem is related to the clockspeed and I know that in the latest version of the Airsim, changes have been made compared to it. (I'm not sure if everything works well)
In fact, a clockspeed of 10 is not really a clockspeed of 10 in NH environment but much more (100 I think). Even a clockspeed of 2 leads to a very strong dynamic of the environment and does not correspond to reality. The car become very instable and the training is complicated because the car takes one or two actions and collides directly. (I decreased the time.sleep respectively and did other changes related to time.)

mitchellspryn, 2018-04-25:
What do you mean by "clockspeed"? I'm not sure I fully understand the issue.

gusnitaa, 2018-04-26:
Clockspeed (https://github.com/Microsoft/AirSim/blob/master/docs/settings.md) : 

"Determines the speed of simulation clock with respect to wall clock. For example, value of 5.0 would mean simulation clock has 5 seconds elapsed when wall clock has 1 second elapsed (i.e. simulation is running faster). The value of 0.1 means that simulation clock is 10X slower than wall clock. The value of 1 means simulation is running in real time. It is important to realize that quality of simulation may decrease as the simulation clock runs faster. You might see artifacts like object moving past obstacles because collison is not detected. However slowing down simulation clock (i.e. values < 1.0) generally improves the quality of simulation."

This function allows to train the model very fast. For example, a clockspeed of 10 allows to train 10 times faster the model. 
Is not really an issue for me, just those who want to use this function, must use the last binaries of Airsim.