microsoft/AutonomousDrivingCookbook #36: Landscape scenario does not show up

bowie2211, 2018-04-24:
## Problem description
Starting AirSim via the provided script AD_Cookbook_Start_AirSim.ps1 with landscape as scenrio ends in a black and white screen.
Starting the other scenrios works fine

## Exepriment/Environment details
* Tutorial used: AirSimE2EDeepLearning
* Environment used: landscape

mitchellspryn, 2018-04-24:
Can you give me the command line that you are using to start it? I just tried running

`
> .\AD_Cookbook_Start_AirSim.ps1 landscape
`

from a powershell window, and everything appears OK.

bowie2211, 2018-04-26:
I have tried to start it from an Power Shell Integrated Console in Visual Code.
Starting with parameter city or coastline works perfectly.
It is only the landscape that makes problems.
I will check out to start it from a Power Shell window outside Visual Code.

mitchellspryn, 2018-04-26:
Intenteresting. Can you also see what happens if you try neighborhood from the Integrated Console?

bowie2211, 2018-04-28:
Neighborhood also is not working.

Could you please give some information how I can run the official AirSim binaries?

mitchellspryn, 2018-04-28:
[Here](https://github.com/Microsoft/AirSim/blob/master/docs/use_precompiled.md). Although, we haven't tested the official binaries with our code, so they're not guaranteed to work. 

What are the specs of the system you are trying to run on?

mitchellspryn, 2018-06-14:
There is no action from us on this issue. Can't reproduce. Closing.

closed 2018-06-14 by mitchellspryn