microsoft/AutonomousDrivingCookbook #37: E2E model failure

andrewazmy, 2018-04-26:
## Problem description
*Trained model has failed to navigate the car through the landscape simulation environment.*

## Problem details
*I have downloaded the AirSim client through the link mentioned in the cookbook readme, along with the training data for the landscape environment. I have followed all notebooks, exploration, train and test. After training for 26 epochs, I have reached a validation loss of 0.00034645, however, the model failed to drive the car on the landscape track simulation, but crashed before it even passed the first bridge.*

## Exepriment/Environment details
* Tutorial used: *AirSimE2EDeepLearning*
* Environment used: *landscape*
* Versions of artifacts used: *Python 3.5.2, Keras 2.1.3*

andrewazmy, 2018-04-27:
I have tried retraining 4 times with almost the same behaviour.

mitchellspryn, 2018-05-10:
That's a bit odd. I re-ran everything this evening, and I was able to get reasonable results. There are a few ways to try to debug:

1) Try to find test images for which the car picks an angle that is really far from the correct answer. See if there are any patterns
2) Try driving the car slower, and see if that has any effect. 

Is the car just driving straight? If so, there might be too many 0 labels in the dataset.

closed 2018-06-08 by mitchellspryn