microsoft/AutonomousDrivingCookbook #38: TrainModel Error

FishMage, 2018-04-29:
## Problem description
*I've completed the data processing and tried to train the model. However, I encountered an error message saying that there is an out of bound error in numpy*

## Problem details
```
D:\AutonomousDrivingCookbook-master\AutonomousDrivingCookbook-master\AirSimE2EDeepLearning>py TrainModel.py

C:\Users\Richard Chen\AppData\Local\Programs\Python\Python36\lib\site-packages\h5py\__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
Using TensorFlow backend.
Successfully init generators
Traceback (most recent call last):
  File "TrainModel.py", line 72, in <module>
    [sample_batch_train_data, sample_batch_test_data] = next(train_generator)
  File "C:\Users\Richard Chen\AppData\Local\Programs\Python\Python36\lib\site-packages\keras\preprocessing\image.py", line 860, in __next__
    return self.next(*args, **kwargs)
  File "D:\AutonomousDrivingCookbook-master\AutonomousDrivingCookbook-master\AirSimE2EDeepLearning\Generator.py", line 242, in next
    return self.__get_indexes(index_array)
  File "D:\AutonomousDrivingCookbook-master\AutonomousDrivingCookbook-master\AirSimE2EDeepLearning\Generator.py", line 264, in __get_indexes
    transformed = self.image_data_generator.random_transform_with_states(x_images.astype(K.floatx()))
  File "D:\AutonomousDrivingCookbook-master\AutonomousDrivingCookbook-master\AirSimE2EDeepLearning\Generator.py", line 140, in random_transform_with_states
    img_channel_axis)
  File "C:\Users\Richard Chen\AppData\Local\Programs\Python\Python36\lib\site-packages\keras\preprocessing\image.py", line 179, in random_channel_shift
    x = np.rollaxis(x, channel_axis, 0)
  File "C:\Users\Richard Chen\AppData\Local\Programs\Python\Python36\lib\site-packages\numpy\core\numeric.py", line 1442, in rollaxis
    axis = normalize_axis_index(axis, n)
numpy.core._internal.AxisError: axis 3 is out of bounds for array of dimension 3
```

## Exepriment/Environment details
* Tutorial used: *\AirSimE2EDeepLearning, I exported a python file using jupyter notebook*
* Environment used: *Windows 10 64-bit*
* Versions of artifacts used (if applicable):  Python 3.6

mitchellspryn, 2018-04-29:
What version of Keras are you using?

mitchellspryn, 2018-04-30:
We've seen [all sorts of issues](https://github.com/Microsoft/AutonomousDrivingCookbook/issues/1) with Keras 2.1.5. Can you try with 2.1.2?

FishMage, 2018-04-30:
That works! Thank you so much!

closed 2018-04-30 by FishMage