microsoft/AutonomousDrivingCookbook #42: TestModel Error

lyhbuaa, 2018-05-10:
**Your issue may already be reported! Please make sure to search all open and closed issues before starting a new one.**

Please fill out the sections below so we can understand your issue better and resolve it quickly.

## Problem description
*I have finished learning the TrainModel.ipynb，and the results looks   good, but when i tried to run the testModel.ipynb, I met this error as the following

## Problem details
*---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-4-abfe90f3be35> in <module>()
----> 1 model = load_model(MODEL_PATH)
      2 
      3 client = CarClient()
      4 client.confirmConnection()
      5 client.enableApiControl(True)

C:\ProgramData\Anaconda3\envs\tensorflow\lib\site-packages\keras\models.py in load_model(filepath, custom_objects, compile)
    232             return custom_objects[obj]
    233         return obj
--> 234     with h5py.File(filepath, mode='r') as f:
    235         # instantiate model
    236         model_config = f.attrs.get('model_config')

C:\ProgramData\Anaconda3\envs\tensorflow\lib\site-packages\h5py\_hl\files.py in __init__(self, name, mode, driver, libver, userblock_size, swmr, **kwds)
    267             with phil:
    268                 fapl = make_fapl(driver, libver, **kwds)
--> 269                 fid = make_fid(name, mode, userblock_size, fapl, swmr=swmr)
    270 
    271                 if swmr_support:

C:\ProgramData\Anaconda3\envs\tensorflow\lib\site-packages\h5py\_hl\files.py in make_fid(name, mode, userblock_size, fapl, fcpl, swmr)
     97         if swmr and swmr_support:
     98             flags |= h5f.ACC_SWMR_READ
---> 99         fid = h5f.open(name, flags, fapl=fapl)
    100     elif mode == 'r+':
    101         fid = h5f.open(name, h5f.ACC_RDWR, fapl=fapl)

h5py\_objects.pyx in h5py._objects.with_phil.wrapper()

h5py\_objects.pyx in h5py._objects.with_phil.wrapper()

h5py\h5f.pyx in h5py.h5f.open()

OSError: Unable to open file (unable to open file: name = 'E:\software\AutonomousDrivingCookbook-master\AirSimE2EDeepLearning\EndToEndLearningRawData\model', errno = 13, error message = 'Permission denied', flags = 0, o_flags = 0)

## Exepriment/Environment details
* Tutorial used: *(AirSimE2EDeepLearning)*
* Environment used: *( landscape)*
* Versions of artifacts used (if applicable): *(Python 3.6, tensorflow 1.6 gpu, Keras 2.1.2)*

mitchellspryn, 2018-05-10:
Do you see any actual model files in the /model/ directory? What is the full file path that is being passed to load_model() in your case? Does it point to a valid file?

lyhbuaa, 2018-05-11:
yes, thank you, there is a folder named models and a （.csv）file named training_log.csv in the model directory, and there are many .h5 files in the .../model/models directory.

And there is another detail, when i import the h5py, It gived me a future warning, so i reinstalled the numpy with 1.3.3 version, and the future warning disappeared, is there any possible this detail has caused the mistake to happen？

mitchellspryn, 2018-05-11:
Not sure about the numpy error, can you paste the full error?

I still don't know the answer to this:

What is the full file path that is being passed to load_model() in your case? Does it point to a valid file?

Also, since you're getting a PermissionsError, can you check the file permissions? Is it read-only? Did you create it as root/admin, and are now trying to access it as a regular user?

lyhbuaa, 2018-05-14:
thank you for your advise, I deleted the modes and runned the TrainModel.ipynb again, Now, the  testModel.ipynb works, but I didn't figure out why the mistake happened, I think may be there is some wrong with the old models. @mitchellspryn

closed 2018-05-14 by mitchellspryn

vighnesh-balaji, 2018-08-09:
The path that i have given to load_model() is MODEL_PATH = 'D:\AutonomousDrivingCookbook-master\AutonomousDrivingCookbook-master\AirSimE2EDeepLearning\model_model.10-0.0002810.h5/'
and I get the same error.

If I don't pass the exact model and just pass the models folder I get an error that says max() argument not found.