microsoft/AutonomousDrivingCookbook #45: TextModel Error

wdsd641417025, 2018-05-22:
**Your issue may already be reported! Please make sure to search all open and closed issues before starting a new one.**

Please fill out the sections below so we can understand your issue better and resolve it quickly.

## Problem description
*I have finished learning the TrainModel.ipynb，and the results looks good, but when i tried to run the testModel.ipynb, I met this error as the following as @llyhbuaa,but it not work when i deleted the modes and runned the TrainModel.ipynb again . I open the Airsim simulink by myself instead of  AD_Cookbook_Start_AirSim.ps1 hawaii which i can't open in powershell.

## Problem details
OSError                                   Traceback (most recent call last)
<ipython-input-3-abfe90f3be35> in <module>()
----> 1 model = load_model(MODEL_PATH)
      2 
      3 client = CarClient()
      4 client.confirmConnection()
      5 client.enableApiControl(True)

C:\ProgramData\Anaconda3\lib\site-packages\keras\models.py in load_model(filepath, custom_objects, compile)
    232             return custom_objects[obj]
    233         return obj
--> 234     with h5py.File(filepath, mode='r') as f:
    235         # instantiate model
    236         model_config = f.attrs.get('model_config')

C:\ProgramData\Anaconda3\lib\site-packages\h5py\_hl\files.py in __init__(self, name, mode, driver, libver, userblock_size, swmr, **kwds)
    267             with phil:
    268                 fapl = make_fapl(driver, libver, **kwds)
--> 269                 fid = make_fid(name, mode, userblock_size, fapl, swmr=swmr)
    270 
    271                 if swmr_support:

C:\ProgramData\Anaconda3\lib\site-packages\h5py\_hl\files.py in make_fid(name, mode, userblock_size, fapl, fcpl, swmr)
     97         if swmr and swmr_support:
     98             flags |= h5f.ACC_SWMR_READ
---> 99         fid = h5f.open(name, flags, fapl=fapl)
    100     elif mode == 'r+':
    101         fid = h5f.open(name, h5f.ACC_RDWR, fapl=fapl)

h5py\_objects.pyx in h5py._objects.with_phil.wrapper()

h5py\_objects.pyx in h5py._objects.with_phil.wrapper()

h5py\h5f.pyx in h5py.h5f.open()

OSError: Unable to open file (unable to open file: name = 'D:\EndToEndDeepLearning\Model\models', errno = 13, error message = 'Permission denied', flags = 0, o_flags = 0)
## Exepriment/Environment details
* Tutorial used: *(AirSimE2EDeepLearning)
* Environment used: *(hawaii )*
* Versions of artifacts used (if applicable): *( Python 3.6, Keras 2.1.2,cntk 2.5.1)*

abc19850508, 2018-05-22:
I saw your error is ”unable to open file: name = 'D:\EndToEndDeepLearning\Model\models'”，I think you have to specify a particular file under the models folder

wdsd641417025, 2018-05-22:
@abc19850508 Thank for your commented , I change 'D:\EndToEndDeepLearning\Model\models' to 'D:\EndToEndDeepLearning\Model',   sadly i doesn't work

mitchellspryn, 2018-05-28:
Is there anything in the models folder? Are you able to read those files (i.e. do you have permission?)

wdsd641417025, 2018-05-30:
@mitchellspryn  thank you yes, thank you, there is a folder named models and a （.csv）file named training_log.csv in the model directory, and there are many .h5 files in the .../model/models directory. And the files can be read

mitchellspryn, 2018-05-31:
For some reason, the file path isn't pointed to the actual model file. From your stack trace:

OSError: Unable to open file (unable to open file: **name = 'D:\EndToEndDeepLearning\Model',** errno = 13, error message = 'Permission denied', flags = 0, o_flags = 0)

This should point to the full model path (e.g. 'D:\EndToEndDeepLearning\Model\Models\12345.h5'). Try passing the full path explicitly to load_model().

wdsd641417025, 2018-06-05:
@mitchellspryn  It does not work either,I run this program in another computer  but have same error

mitchellspryn, 2018-06-06:
What are the contents of the string being passed to load_model? Does it point to a valid model on disk? If no, change the string to point to a valid model on disk and try again.

mitchellspryn, 2018-06-14:
@wdsd641417025 : Did you get it to work? Did you find out what was being passed to load_model?

lwluowei, 2018-07-08:
dada

lwluowei, 2018-07-08:
Same problem here,
win 10 OS 
[image] 
Can not change the read-only property,i am not sure it is the reason  of the problem.

lwluowei, 2018-07-08:
Found the problem!
[image]
I replace the None with the needed model path which is not supposed to,Follow the picture setting will close the problem.

mitchellspryn, 2018-07-08:
Odd that you couldn't change the read-only property of the models (or the fact that it was set in the first place). Did you run the training process as administrator?

Anyway, it looks like the issue is solved, so I'll close this. Thanks for posting detailed instructions!

closed 2018-07-08 by mitchellspryn

LouisVann, 2020-11-01:
Hey I met the same problem. I don't know what happend, but when I changed `load_model` which is imported before to `keras.models.load_model`, everything changed!!!

[image]

So who can tell me why?

LouisVann, 2020-11-02:
> Hey I met the same problem. I don't know what happend, but when I changed `load_model` which is imported before to `keras.models.load_model`, everything changed!!!
> 
> [image]
> 
> So who can tell me why?

And I find that function `load_model` which is imported before uses actually `keras.engine.saving.load_model`
while `keras.models.load_model` uses `tensorflow_core.python.keras.saving.save.load_model`.
Maybe thw two have different permissions?

LouisVann, 2020-11-02:
> > Hey I met the same problem. I don't know what happend, but when I changed `load_model` which was imported before to `keras.models.load_model`, everything changed!!!
> > [image]
> > So who can tell me why?
> 
> And I find that function `load_model` which was imported before uses actually `keras.engine.saving.load_model`
> while `keras.models.load_model` uses `tensorflow_core.python.keras.saving.save.load_model`.
> Maybe the two have different permissions?

But why could the two have different references?

camartin88, 2020-11-17:
"> But why could the two have different references?"

Just had the same above problem and solved it this way. I called a friend and he explained if you use "from ____ import ____" it will only import the one function, however if that function relies on any other dependencies with in your original ____  using the dot (my.werid.function) notation allows python to search upwards for those dependencies.