microsoft/AutonomousDrivingCookbook #46: Errors in train generator

CMXB, 2018-05-24:
**Your issue may already be reported! Please make sure to search all open and closed issues before starting a new one.**

Please fill out the sections below so we can understand your issue better and resolve it quickly.

## Problem description
[sample_batch_train_data, sample_batch_test_data] = next(train_generator)  
this line in TrainModel results in an axis error,  and I can't figure this out

## Problem details
---------------------------------------------------------------------------
AxisError                                 Traceback (most recent call last)
<ipython-input-4-6955bc10ff72> in <module>()
     24     plt.show()
     25 
---> 26 [sample_batch_train_data, sample_batch_test_data] = next(train_generator)
     27 for i in range(0, 3, 1):
     28     draw_image_with_label(sample_batch_train_data[0][i], sample_batch_test_data[i])

E:\software\anaconda\lib\site-packages\keras\preprocessing\image.py in __next__(self, *args, **kwargs)
   1067 
   1068     def __next__(self, *args, **kwargs):
-> 1069         return self.next(*args, **kwargs)
   1070 
   1071     def _get_batches_of_transformed_samples(self, index_array):

E:\Projects\Autonomous driving\Code\jupyter\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in next(self)
    240         # so it can be done in parallel
    241 
--> 242         return self.__get_indexes(index_array)
    243 
    244     def __get_indexes(self, index_array):

E:\Projects\Autonomous driving\Code\jupyter\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in __get_indexes(self, index_array)
    262                 x_images = x_images[self.roi[0]:self.roi[1], self.roi[2]:self.roi[3], :]
    263 
--> 264             transformed = self.image_data_generator.random_transform_with_states(x_images.astype(K.floatx()))
    265             x_images = transformed[0]
    266             is_horiz_flipped.append(transformed[1])

E:\Projects\Autonomous driving\Code\jupyter\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in random_transform_with_states(self, x, seed)
    138             x = image.random_channel_shift(x,
    139                                      self.channel_shift_range,
--> 140                                      img_channel_axis)
    141         if self.horizontal_flip:
    142             if np.random.random() < 0.5:

E:\software\anaconda\lib\site-packages\keras\preprocessing\image.py in random_channel_shift(x, intensity, channel_axis)
    188 
    189     """
--> 190     x = np.rollaxis(x, channel_axis, 0)
    191     min_x, max_x = np.min(x), np.max(x)
    192     channel_images = [np.clip(x_channel + np.random.uniform(-intensity, intensity), min_x, max_x)

E:\software\anaconda\lib\site-packages\numpy\core\numeric.py in rollaxis(a, axis, start)
   1440     """
   1441     n = a.ndim
-> 1442     axis = normalize_axis_index(axis, n)
   1443     if start < 0:
   1444         start += n

AxisError: axis 3 is out of bounds for array of dimension 3

## Exepriment/Environment details
* Tutorial used:  AirSimE2EDeepLearning
* Environment used:  landscape

CMXB, 2018-05-24:
I forget to look the closed issue, please close this

closed 2018-05-25 by adshar

Sunshinehualong, 2018-05-31:
I  also got this problem, did you have figured this out?