microsoft/AutonomousDrivingCookbook #47: xbox one controller with Hawaii

mnpica, 2018-06-04:
Thanks for writing this tutorial!
But there is a question about the xbox controller.

1.Using the scene "AD_Cookbook_AirSim\Hawaii" with xbox one controller , the input steering and throttle value are too large. How to set it properly?
[image]

2.When problem 1 happened, pressing backspace  would not make all values back to 0 . The car will keep rotating.

mnpica, 2018-06-04:
This problem doesn't happen in those five environments.
After I finished the "Autonomous Driving using End-to-End Deep Learning: an AirSim tutorial", I want to collect data manually because I don't know how to clone the car behavior.
I can't go backwards with xbox controller in 'Neighbourhood' .There are some cars and shadows on the road making lots of collisions, and so does 'Mountain Landscape.'
I decided to collect data in 'Hawaii' with keyboard.  The trained model isn't very well cause the steering angel was -1 ,0 ,or 1 only.
I would appreciate it if you have any advice for me.

mitchellspryn, 2018-06-04:
I'm still a bit unclear; what do you mean by "clone the car behavior?"

If you are trying to train a new end-to-end model for the Hawaii environment by manually collecting data, then you should be OK using the official airsim binary. Just record your driving using the record button, and drive around for a few hours. Then, you should be able to use the code in the tutorial to train the end to end model on that dataset, and then test it in the Hawaii environment.

mnpica, 2018-06-04:
1.
I've already collected data manually by using record button. The steering angle is -1 ,0 or 1 if I use keyboard for input.  So I want to know how to setting the xbox controller or steering wheel.
2.
The tutorials seem to have two well-trained models to collect data from API. I thought the "behavior cloning" is that let the computer automatically generate data.  I'm new to this , so I can't do it by now.

mitchellspryn, 2018-06-04:
Collecting with the keyboard will give you only 1,0,-1 labels because a key is only down or up. If you use the controller or the steering wheel, you will get continuous values. You may look into using something like [x360ce](https://www.x360ce.com/) if it doesn't work out of the box. 

I do not understand what you are saying with the second point. The E2E tutorial trains a model based on the data provided. There is no "generation" of data. If you change the data, you will get a different model. I'm suggesting that you might have success by changing the dataset from ours (which was collected on the Mountain environment) with a dataset collected from the Hawaii environment.

mnpica, 2018-06-04:
Thanks for advise.
I'll working on x360ce with Hawaii environment to check whether there is same problem or not.
If there is a fix for this environment with steering wheel or xbox controller, please let me know. Thank you!

mitchellspryn, 2018-06-14:
@mnpica : Does this work with the official builds downloaded from the airsim repo?

mitchellspryn, 2018-08-21:
Closing due to inactivity.

closed 2018-08-21 by mitchellspryn