microsoft/AutonomousDrivingCookbook #51: Numpy Error, Help me

b3rg0n, 2018-06-08:
TypeError: bad operand type for unary -: 'str'

python version = 3.6
Numpy version = 1.14.3
scipy version = 1.10

Help me... 
**Can I know which version I used at the time of implementation?**

mitchellspryn, 2018-06-08:
Please use the template. It's there for a reason - it'll help you provide us with the information we need to debug your isuse. 

So, the typeerror most likely means that you've tried to perform a - operation on a string. I don't know why that's happening because I don't have the location where the error happened. Can you give me the following:

* Full stack trace where your error occured
* Tutorial that you are trying to do
* The keras and tensorflow versions you are using (in fact, the full output of 'pip freeze' might be helpful)
* Any code changes you have made. If you've made any, try re-running from source.

mitchellspryn, 2018-06-08:
Is [this issue](https://github.com/Microsoft/AutonomousDrivingCookbook/issues/50) the same thing? If so, please don't open multiple issues for the same problem.

I've looked at that stack trace, and it's failing within the keras image preprocessing library:

C:\ProgramData\Anaconda3\lib\site-packages\keras_preprocessing\image.py in random_channel_shift(x, intensity_range, channel_axis)
191 Numpy image tensor.
192 """
--> 193 intensity = np.random.uniform(-intensity_range, intensity_range)
194 return apply_channel_shift(x, intensity, channel_axis=channel_axis)
195

Tracing back up, it looks like the channel_shift_range passed to DriveDataGenerator is the wrong type. Did you change this parameter? If so, what value did you set it to? Setting it to 0 will skip the image.random_channel_shift() call, so that might be a quick fix to try.

b3rg0n, 2018-06-14:
Tanks you, 
Looking for other people's answers, I could see that the Keras version was a problem. I reinstalled Keras vsersion 2.1.2 and solved this problem.

mitchellspryn, 2018-06-14:
Oh. Good to know. I'll close this, then.

closed 2018-06-14 by mitchellspryn

MegaWind, 2018-08-19:
Hi - here's the issue that is causing the tutorial to break in Keras 2.1.5+ with the error noted above, specifically:
"C:\ProgramData\Anaconda3\lib\site-packages\keras_preprocessing\image.py in random_channel_shift(x, intensity_range, channel_axis)
191 Numpy image tensor.
192 """
--> 193 intensity = np.random.uniform(-intensity_range, intensity_range)"
**TypeError: bad operand type for unary -: 'str'**

In Keras 2.1.5, they added a new init parameter brightness_range for the ImageDataGenerator -  "Add brightness_range data augmentation argument in keras.preprocessing.image.ImageDataGenerator."

Class DriveDataGenerator is thus one param short - so the parameters are forced back:
                 height_shift_range=0.,
                 **brightness_range=(0,.4),** #<--- new param for 2.1.5
                 shear_range=0.,
                 zoom_range=0.,
                 **channel_shift_range=0.,
                 fill_mode='nearest',**

Thus, channel_shift_range is pushed back in terms of param mapping (thanks to Pythons dynamic typing no error is shown), and has value "nearest", a string, by taking on fill_mode's value. 
 Thus channel_shift_range is now a string instead of a float and the subsequent typedef operation error when negation is attempted.
I was able to fix by adding in the missing param brightness range in the class __init__ function which restores the parameter mapping for initialization.   However, doing so would also break it for Keras 2.1.4 and below...so not something I will check in but wanted to show the issue.  Note that InstallPackages.py does not enforce a Keras version so people will all hit this issue with 2.1.5+

For now though the easy fix for anyone hitting this:
pip uninstall keras
pip install keras==2.1.2  (or anything < 2.1.5 really)
:)

DanielDuck, 2018-08-24:
Hi, I'm having the same issue.

I tried to do what MegaWind suggested:
`pip uninstall keras`
`pip install keras==2.1.2`

 But this caused incompatibility problems with the keras packages.

> keras-preprocessing 1.0.1 has requirement keras>=2.1.6, but you'll have keras 2.1.2 which is incompatible.
> keras-applications 1.0.4 has requirement keras>=2.1.6, but you'll have keras 2.1.2 which is incompatible.

Any suggestion of what I could do?

mitchellspryn, 2018-08-24:
Uninstalling the keras-preprocessing and keras-applications packages might help. You may consider using a venv specifically for these notebooks if you need these packages for other applications.

MegaWind, 2018-08-24:
[Generator_py_fix.txt](https://github.com/Microsoft/AutonomousDrivingCookbook/files/2316872/Generator_py_fix.txt)

@DanielDuck  - you can backup your current Generator.py file and then save this one as a new Generator.py.
That will resolve for  2.1.5+ Keras.  Don't go too far though as I hit a different issue after this was fixed using latest Keras (2.2.2) and didn't feel like debugging.  2.1.6 seemed to be fine though.   (note - I have Keras preprocessing 1.0.2 and applications 1.0.4 and didn't get the error you did which is odd running Keras 2.1.6 but the new Generator.py above will resolve).

@mitchellspryn  - why not just fix the parameters as noted, and make a new drop for the tutorial since this was built on Dec 17 Keras (i.e. half a year ago)?  People aren't going to want to downgrade Keras just to use the tutorial and may not bother trying to find an answer if they crash working through the tutorial.  I think that would avoid a lot of people giving up and moving on re; tutorial.

DanielDuck, 2018-08-25:
@MegaWind 

I really appreciate your help.
I got Keras 2.1.6, Keras preprocessing 1.0.2 and applications 1.0.4 and saved "Generator_py_fix.txt" as a new "Generator.py".

But I keep getting the same error. Any idea of what else can be wrong?

MegaWind, 2018-08-25:
@DanielDuck 
Hi that should have fixed the issue.   But one possibility is if you still have old values loaded in your notebook?
  Assuming you are running in the Juptyer notebook - you did a full kernel restart (the upper toolbar button that warns you all variables will be reset) and then walked through the sections again?  Otherwise if you installed but didn't restart the notebook it would still have the old variables loaded and you would continue to get the same error.
If that's not it let me know!

DanielDuck, 2018-08-25:
You are right!

I did a full kernel restart and it worked!
I did walk through the sections again before restarting. I thought that would be enough to rewrite the variables, but somehow it didn't.

Anyway, thank you very much for your help!

MegaWind, 2018-08-25:
@DanielDuck  - awesome, glad to hear that was the issue and you are able to continue the tutorial.
Thanks for letting me know it is working for you now!

mitchellspryn, 2018-08-25:
Thanks @MegaWind for helping :)