microsoft/AutonomousDrivingCookbook #53: City environment issues

b3rg0n, 2018-06-12:
## Problem description
**1) City environment steering problem**

**2) Deep Learning Learning Outcomes Optimization Problem**

## Problem details
_1) Run x360e.exe in .. \ AD_Cookbook_AirSim \ City \ CityEnviron \ Binaries \ Win64. Then, the steering is freely input in the game, and it is not operated. How can I solve this problem?

2) It runs well in a **neighborhood environment**. After learning the deep learning, the car moves very zigzag. Is there any way to optimize it for normal driving?_

## Exepriment/Environment details
* Tutorial used: *AirSimE2EDeepLearning*
* Environment used: *1) city, neighborhood *
* Versions of artifacts used (if applicable): *Python 3.6, Keras 2.1.2, Cuda9.0, Xbox One controller, GTX960*

mitchellspryn, 2018-06-14:
I don't understand the issue in (1). Is the car not responding to the xbox controller? Is the car responding, but doing the wrong thing?

Also, I think the latest builds produced on the [airsim repo](https://github.com/Microsoft/AirSim/releases) support controllers natively, so you may have luck trying that build. 

2) What is the plot that you shared? There are ways to change the behavior of the model - change the architecture, change the preprocessing steps. If the car is swerving too much, think about which of our preprocessing steps might address that and try changing the parameters. If you're still stuck, feel free to email me at mspryn@microsoft.com, and we can continue the discussion - I don't want to post too many analysis answers for fear that others will just read the answer instead of thinking about and understanding the solution.

b3rg0n, 2018-06-14:
답변 주셔서 감사합니다. 

1번 문제는 새로운 빌드 버전으로 시도해보겠다! 
위의 질문들을 상세히 적어서, 메일로 따로 드리겠습니다!

mitchellspryn, 2018-08-21:
Closing due to inactivity.

closed 2018-08-21 by mitchellspryn