microsoft/AutonomousDrivingCookbook #54: Steering range of the car in AirSimE2EDeepLearning

mnpica, 2018-06-13:
## Problem description
What is the steering range of the default avatar car? 

## Problem details
1.
In "AirSimE2EDeepLearning/TrainModel.ipynb" , I found a comment for the steering range is +- 40 degees.
*(theta = label * 0.69 #Steering range for the car is +- 40 degrees -> 0.69 radians)*
But in "TestModel.ipynb",  the steering range is 0.5 radians(28.65 degrees). Which one is right?
*(car_controls.steering = round(0.5 * float(model_output[0][0]), 2))*
2.
How can I get the details of avatar car ? Can I change the settings of the avatar car?
Is possible to use a custom car to run this project?
3.
The gear data  is automatically changed according the speed data. Should we delete this column in 'airsim_rec.txt' since we don't need to predict the gear or use this data to predict?
4.
Using the record button, I can get the data from one camera  in RGB mode. How to record the data from other cameras or other modes with record button? 

## Exepriment/Environment details
* Tutorial used: AirSimE2EDeepLearning
* Environment used:  hawaii
* Versions of artifacts used (if applicable): Python 3.6.5, Keras 2.1.2

mitchellspryn, 2018-06-13:
1) The comment in the Train.ipynb. See answer to (2).

2) The 0.5 in that expression is not 0.5 radians. When testing, I found that the model had a tendency to overcorrect and oversteer. So I added that constant to smooth the output of the model. That should give you an idea of why setting this constant to 0.0009 would cause poor performance. Do note that I would expect that you'd have to retrain the model for Hawaii, which means collecting data for that environment.

3) What do you mean "details"? You can change all of the vehicle model parameters if you download and build AirSim from source. The car asset files are included in the plugin. You'd probably need to recollect data if you change the parameters of the car.

4) You can. It shouldn't have any effect on the modeling, apart from potentially reducing your file size by a bit. 

5) You can't without modifying the AirSim source code. To record other views, and more complex scenarios, you can use the python or c++ apis. The build of AirSim that we include with the tutorial is before this feature was introduced.

mnpica, 2018-06-13:
Thanks! I'll try to built the AirSim by my own with LandscapeMountain.
One last question is the environments (NH,city,Hawaii...etc.) are created by your team?
In marketplace of EPIC GAMES, there are few road environments suited for autonomous car.

mitchellspryn, 2018-06-13:
Neighborhood and Landscape are downloaded from the Unreal store. Hawaii and City are built by the AirSim team; the assets for those are not currently available for download.

closed 2018-06-13 by mitchellspryn

Misslindalian, 2019-06-04:
Hello, I have not found your answer about the range of steering angles. Can you tell me again? And how is the label of the steering angle normalized to [-1, 1]? Is there a formula or reference?

damilolah, 2019-10-24:
Please help!!!

I have questions relating to gear in AirSim document.

What attribute in AirSim_rec indicates when the vehicle is backing up or in reverse?
What gear value indicates reverse
Can we predict forward/reverse movement along side steering angle to allow the vehicle correct itself in case it hits an obstacle.
Thank you in anticipation.