microsoft/AutonomousDrivingCookbook #55: data preprocessing question

yluo19, 2018-06-13:
I have tried to run tutorial (set brighten_range  = 0.4, drop 0.9 data that label = 0) but find the result was not good like vehicle have collision on downhill path.
Then I have tried only drop 0.5 dataset which label = 0 and leave brighten_range = 0.4 , got better result than before. The vehicle was slip down when pass the shadow region so I though that maybe brighten_range could affect result. So I tried to run with brighten_range = 0, zero_drop_percentage = 0. I realized the vehicle runs on the left side of road and slip down when climbing. result worse. 

## Question:
1. For Brighten_Range parameter, how does this parameter affect the result? Why did you improve the brightness up to 40%?
2.. ROI: [76,135,0,255], Zero_Drop_Percentage, Brighten_Range. Does these three parameters have some relations between each other? I want to try some different values in order to find the best parameters value.

mitchellspryn, 2018-06-14:
1) Did you visualize images with different values of this parameter for the transform?

2) I suggest reviewing the comments in TrainModel.ipynb - it clearly labels what each of these transform parameters do. If that's not clear enough, you can read the code in generator.py to see exactly what the effect of changing each of these parameters are. It is also helpful to visualize images after they have been processed to see the effects.

yluo19, 2018-06-14:
Thank for answering! I tried different values of parameter above. I found car always can not pass  a shadow region so I think that brighten_range affect this.  I am trying different values of brighten_range but I have no idea why brightness need to be improved or changed?

mitchellspryn, 2018-08-21:
Try visualizing the image before and after the brighten_range transform. It should become obvious what that transform does.

closed 2018-08-21 by mitchellspryn