microsoft/AutonomousDrivingCookbook #56: Launch local training job in DistributedRL

e271141, 2018-06-29:
I launch the train.bat for Reinforcement learning. The car may occasionally tumbling on the ground or rotating in air for each new epoch. I tried to fix this by modify the distributed_agent.py. In function __get_next_starting_point() I set "random_direction=(0,0,0)", but it doesn't work.  What is the cause of this problem?

## Exepriment/Environment details
* Tutorial used: AirSimE2EDeepLearning, DistributedRL
* Environment used: Neighborhood

e271141, 2018-07-02:
Ok, I figure it out myself. As it turns out, the Z direction points down the surface. We should set the starting point as -1 or some minus number to make sure the car won't get stuck in the road.

closed 2018-07-02 by e271141

Curry30h, 2019-03-27:
hello，can you tell me how should I modify the code in distributed_agent.py so that it can be trained locally? Thank you.