microsoft/AutonomousDrivingCookbook #58: transport error!help

lwluowei, 2018-07-08:
**Your issue may already be reported! Please make sure to search all open and closed issues before starting a new one.**

Please fill out the sections below so we can understand your issue better and resolve it quickly.

## Problem description
*(Please provide a 2-3 sentence description of your problem. Be concise to ensure this description is useful for future users who might run into the same issue.)*

## Problem details
*(Please describe your problem in as much detail as possible here. Make sure to include screenshots, code snippets, error messages, links and anything else you think will help us understand your problem better. If applicable, please also provide us a list of steps to reproduce your problem.)*

## Exepriment/Environment details
* Tutorial used: *(For example, AirSimE2EDeepLearning, DistributedRL etc.)*
* Environment used: *(For example, landscape, city, hawaii etc.)*
* Versions of artifacts used (if applicable): *(For example, Python 3.5, Keras 2.1.2 etc.)*
[image]
Have no idea what to do with this kind of problem:<

mitchellspryn, 2018-07-08:
This means it can't connect to AirSim, for whatever reason. Did you start the exe?

lwluowei, 2018-07-09:
You get it right,I started with the wrong scenario .Thank U so much, finally I got it worked,it is funny!

closed 2018-08-21 by mitchellspryn