microsoft/AutonomousDrivingCookbook #60: getHomeGeoPoint' was called with an invalid number of arguments

mckygit, 2018-08-10:
**Your issue may already be reported! Please make sure to search all open and closed issues before starting a new one.**

Please fill out the sections below so we can understand your issue better and resolve it quickly.

## Problem description

In TestModel.ipynb, I get the error message below when running the code.

model = load_model(MODEL_PATH)

client = CarClient()
client.confirmConnection()
client.enableApiControl(True)
car_controls = CarControls()
print('Connection established!')

I have changed the port to 42451 (from 41452) in the AirSimClient.py to address the "Waiting for connection... Retry connection over the limit" issue , https://github.com/Microsoft/AutonomousDrivingCookbook/issues/12

## Problem details
WARNING:tensorflow:From c:\users\mteng\anaconda3\envs\tensorflow\lib\site-packages\keras\backend\tensorflow_backend.py:1264: calling reduce_prod (from tensorflow.python.ops.math_ops) with keep_dims is deprecated and will be removed in a future version.
Instructions for updating:
keep_dims is deprecated, use keepdims instead
WARNING:tensorflow:From c:\users\mteng\anaconda3\envs\tensorflow\lib\site-packages\keras\backend\tensorflow_backend.py:1349: calling reduce_mean (from tensorflow.python.ops.math_ops) with keep_dims is deprecated and will be removed in a future version.
Instructions for updating:
keep_dims is deprecated, use keepdims instead
Waiting for connection: 

---------------------------------------------------------------------------
RPCError                                  Traceback (most recent call last)
<ipython-input-2-abfe90f3be35> in <module>()
      2 
      3 client = CarClient()
----> 4 client.confirmConnection()
      5 client.enableApiControl(True)
      6 car_controls = CarControls()

D:\Home\Public\local\workspace\jupyter\GitHub\AutonomousDrivingCookbook\AirSimE2EDeepLearning\AirSimClient.py in confirmConnection(self)
    158     def confirmConnection(self):
    159         print('Waiting for connection: ', end='')
--> 160         home = self.getHomeGeoPoint()
    161         while ((home.latitude == 0 and home.longitude == 0 and home.altitude == 0) or
    162                 math.isnan(home.latitude) or  math.isnan(home.longitude) or  math.isnan(home.altitude)):

D:\Home\Public\local\workspace\jupyter\GitHub\AutonomousDrivingCookbook\AirSimE2EDeepLearning\AirSimClient.py in getHomeGeoPoint(self)
    167 
    168     def getHomeGeoPoint(self):
--> 169         return GeoPoint.from_msgpack(self.client.call('getHomeGeoPoint'))
    170 
    171     # basic flight control

c:\users\mteng\anaconda3\envs\tensorflow\lib\site-packages\msgpackrpc\session.py in call(self, method, *args)
     39 
     40     def call(self, method, *args):
---> 41         return self.send_request(method, args).get()
     42 
     43     def call_async(self, method, *args):

c:\users\mteng\anaconda3\envs\tensorflow\lib\site-packages\msgpackrpc\future.py in get(self)
     43                         raise self._error
     44                     else:
---> 45                         raise error.RPCError(self._error)
     46             else:
     47                 return self._result

RPCError: b"rpclib: client error C0002: Function 'getHomeGeoPoint' was called with an invalid number of arguments. Expected: 0, got: 1"

## Exepriment/Environment details
* Tutorial used: *(For example, AirSimE2EDeepLearning, DistributedRL etc.)*
AirSimE2EDeepLearning
* Environment used: *(For example, landscape, city, hawaii etc.)*
LandscapeMountians
* Versions of artifacts used (if applicable): *(For example, Python 3.5, Keras 2.1.2 etc.)*
Keras               2.1.2
numpy               1.15.0

mitchellspryn, 2018-08-10:
Are you using the airsim binaries we provided or the ones from the AirSim page? If you aren't using ours, please try with that.

mckygit, 2018-08-10:
I am using v1.2.0 - Windows landscapemountains.zip from https://github.com/Microsoft/AirSim/releases

if I use the simulator package from here, https://github.com/Microsoft/AutonomousDrivingCookbook/tree/master/AirSimE2EDeepLearning I get another different error.

"
[image]

mckygit, 2018-08-10:
I moved the folder to c: drive and now  ".\AD_Cookbook_Start_AirSim.ps1 landscape " seems to run.   I was previously running it from d drive which resulted in the fatal error above.   

This also fixed the original  "invalid number of arguments" error.

What is the difference between the airsim and ad_cookbool_start_airsim binaries besides having different default listening ports?

mitchellspryn, 2018-08-21:
See here: https://github.com/Microsoft/AutonomousDrivingCookbook/issues/62. Short answer - no difference, it's just an older version of the AirSim plugin that we've kept for API compatibility.

closed 2018-08-21 by mitchellspryn