microsoft/AutonomousDrivingCookbook #61: Shadows causing issue for algorithm

NextSim, 2018-08-10:
## Problem description
Shadows in the neighborhood environment seem to be causing trouble for the autonomous algorithm

## Problem details
We are using reinforced learning with air sim to teach a car to drive through the neighborhood environment.  Overall it seems to be learning well and is getting better at driving.  It seems though that when it drives through shadows (like the ones cast by the trees) the camera can no longer see and it frequently turns and drives off the road.

## Exepriment/Environment details
* Tutorial used: DistributedRL
* Environment used: Neighborhood

mitchellspryn, 2018-08-21:
Yup! That's a good observation. There are many strategies you can try to deal with this issue. One that seems popular is during training, modify the brightness of random patches of the input image to simulate additional "shadows," to make the model more robust to lighting. 

Ultimately, there is no easy answer, you'll have to experiment and see what works.

closed 2018-08-21 by mitchellspryn

NextSim, 2018-08-21:
ok great.. thanks!