microsoft/AutonomousDrivingCookbook #63: max() arg is an empty sequence (no model file)

oyhq0708, 2018-08-21:
**Your issue may already be reported! Please make sure to search all open and closed issues before starting a new one.**

Please fill out the sections below so we can understand your issue better and resolve it quickly.

## Problem description
while I run the TestModel, it gets stuck at line 17 saying "ValueError: max() arg is an empty sequence". I try to solve it by searching h5 files in the model/models but I can't even find the model folder. 

## Problem details

Copy of the error message below:
ValueError                                Traceback (most recent call last)
<ipython-input-1-f4310083b67f> in <module>()
     15 if (MODEL_PATH == None):
     16     models = glob.glob('model/models/*.h5')
---> 17     best_model = max(models, key=os.path.getctime)
     18     MODEL_PATH = best_model
     19 

ValueError: max() arg is an empty sequence

## Exepriment/Environment details
* Tutorial used: AirSimE2EDeepLearning
* Environment used: landscape
* Versions of artifacts used (if applicable): Python 3.6, Keras 2.2.0

mitchellspryn, 2018-08-21:
1) Please use Keras 2.1.2
2) Did you complete running TrainModel? If so, there should be a models folder in the directory you are running the notebook from. Try hardcoding MODEL_PATH to point to the best performing model in that directory.

oyhq0708, 2018-08-21:
Oh no I didn't. When I just ran TrainModel, it raises another issue:
"
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-9-54951b8e4010> in <module>()
     14 import os
     15 import numpy as np
---> 16 import pandas as pd
     17 from Generator import DriveDataGenerator
     18 from Cooking import checkAndCreateDir

C:\ProgramData\Miniconda3\lib\site-packages\pandas\__init__.py in <module>()
     40 import pandas.core.config_init
     41 
---> 42 from pandas.core.api import *
     43 from pandas.core.sparse.api import *
     44 from pandas.tseries.api import *

C:\ProgramData\Miniconda3\lib\site-packages\pandas\core\api.py in <module>()
      8 from pandas.core.dtypes.missing import isna, isnull, notna, notnull
      9 from pandas.core.arrays import Categorical
---> 10 from pandas.core.groupby.groupby import Grouper
     11 from pandas.io.formats.format import set_eng_float_format
     12 from pandas.core.index import (Index, CategoricalIndex, Int64Index,

C:\ProgramData\Miniconda3\lib\site-packages\pandas\core\groupby\__init__.py in <module>()
      1 # flake8: noqa
----> 2 from pandas.core.groupby.groupby import (
      3     Grouper, GroupBy, SeriesGroupBy, DataFrameGroupBy
      4 )

C:\ProgramData\Miniconda3\lib\site-packages\pandas\core\groupby\groupby.py in <module>()
     47                                CategoricalIndex, _ensure_index)
     48 from pandas.core.arrays import ExtensionArray, Categorical
---> 49 from pandas.core.frame import DataFrame
     50 from pandas.core.generic import NDFrame, _shared_docs
     51 from pandas.core.internals import BlockManager, make_block

C:\ProgramData\Miniconda3\lib\site-packages\pandas\core\frame.py in <module>()
     72                                    create_block_manager_from_arrays,
     73                                    create_block_manager_from_blocks)
---> 74 from pandas.core.series import Series
     75 from pandas.core.arrays import Categorical, ExtensionArray
     76 import pandas.core.algorithms as algorithms

C:\ProgramData\Miniconda3\lib\site-packages\pandas\core\series.py in <module>()
     65 from pandas.compat.numpy import function as nv
     66 
---> 67 import pandas.core.ops as ops
     68 import pandas.core.algorithms as algorithms
     69 

AttributeError: module 'pandas' has no attribute 'core'
"
I tried to install packages again but it still has the same error as above.

oyhq0708, 2018-08-22:
I solved it by downgrading pandas version. Thanks for the help

closed 2018-08-22 by oyhq0708