microsoft/AutonomousDrivingCookbook #66: Speeding Up the Training

Gaduu, 2018-08-24:
Hi,

Just going through the Distributed RL Example. Basically, I have only just started working with AirSim itself, I was wondering if you can possibly train the model without rendering the simulation? Secondly, my rig consists of 2x1080's so would be great if I could utilise both GPU's but as far as I can tell the sample project uses only one. Is there any way to support multi-gpu configuration?

Cheers,

mitchellspryn, 2018-08-25:
To your first point: even if you could, it'd be unlikely that it would speed things up. Ultimately, the unreal engine would still have to render the image, so it would still have to draw most of the scene. During our training on Azure, we actually did not have the visual window running (due to some details about how Batch starts the training process). We did not observe a speed increase due to not rendering the scene. 

Regarding the multiple GPUs: It's unlikely that a single training instance would benefit from multiple GPUs. The model itself is quite small, so it easily fits inside a single GPU. Also, the bottleneck isn't training the model, it's generating the data. So even if you configured the tensorflow session to use both GPUs, you wouldn't see any benefit.

What would potentially be interesting is starting multiple agent sessions on the same machine. In this case, it would be similar to the scenario we had in batch in which multiple agents were running in parallel, and communicating the weight updates to the parameter server. In your scenario, you would have two agents running and one parameter server. You'd have to modify the code to conditionally use the different devices (probably depending on a command line arg) for the agents, but it could lead to a ~2x speedup, if your CPU doesn't become the bottleneck.

Gaduu, 2018-08-28:
Hi,

Great that explains a lot, the multiple-agent is probably an objective to go for right now, having multiple agents collecting different observations should probably do a lot of good. 

One more question regarding the code, so I've noticed that the model is saved to .json files, I have been training the model from the scratch for few days now, just really testing this thing and seeing how it works, just to confirm if I stop the training and want to resume it can I just use the .json file to load the model/weights and continue from the point I have stopped?

Cheers,

mitchellspryn, 2019-01-30:
Sorry, didn't see your comment. 

Yes. You can use the .json file to reload the weights to continue training.

closed 2019-01-30 by mitchellspryn