microsoft/AutonomousDrivingCookbook #67: Has anyone updated the python scripts to use the latest API?

NextSim, 2018-08-24:
## Problem description
Update python code to use latest API

## Problem details
I know that the tutorials won't be updated to use the new API, at least not for a while, but I'm wondering if anyone has done this on their own.  I have gotten part way through updating it, but when I run train.bat now the car doesn't start on the road, it starts anywhere on the map, or sometimes in the sky.

It looks like the road lines and reward points are being read in ok.. so I'm guessing it's an issue with how the position is sent to airsim.. just wondering if anyone else has gone down this path.

## Exepriment/Environment details
* Tutorial used:   DistributedRL*
* Environment used: neighborhood

NextSim, 2018-08-27:
Ok... that makes sense. I thought something like that may be going on, so I went through on the latest map and re-mapped out the road lines and reward points, by driving to each location and then using the getCarState API to get the position, but I still see the same behavior.. I'll see if I can figure out the starting point offset.  

Thanks

closed 2018-09-04 by NextSim

wonjoonSeol, 2019-03-14:
@NextSim, I have nearly finished updating the code to work with the latest Airsim binary.
Do you still have that roadlines and rewards points txt?

NextSim, 2019-03-14:
No I don't.  I did get it working, the key piece is to make sure that the points are all aligned on straight lines, and there are no slight offsets.  

Plotting in excel helped to see this.