microsoft/AutonomousDrivingCookbook #70: Why does the vehicle speed seem to be unchanged after training.

BrianN92, 2018-09-20:
Hi, I was following the tutorial of E2E deep learning for autonomous driving. Now the vehicle is able to successfully run by itself after training. However, the speed of the vehicle seemed to be unchanged, which was always around 5. Does the model really take the vehicle speed as a feature during training? Did anyone meet this problem? Thanks!

adshar, 2018-09-20:
Hi @BrianN92 , thanks for trying out the tutorial. The car speed is indeed part of the `previous_state` tuple which contains: (steering, throttle, brake, speed) values as the last known state of the system. This tuple is fed into the training network as a feature vector along with image features extracted from convolutional layers. The prediction from the model however, is only for the steering angle. You can confirm this by making note of the single neuron in the output layer of the network: 
`merged = Dense(1, name='output')(merged)`

The model is predicting the steering angle based on visual input through the camera as well as awareness of its last known state comprising of the steering, throttle, brake and speed values. When you run the trained model there is a PID controller that helps maintain constant speed as your model takes control of the steering wheel. 

You are most definitely encouraged to play around with the tutorial to predict more than one output (you will have to make some minor edits to the Cooking.py file, around that line 95 for loop). Another thing to note if you do decide to predict more than just the steering angle, you should not be predicting speed, but rather the throttle. It's just like driving in real life where you don't control the speed of your car directly, you control how hard you press the gas pedal instead.

BrianN92, 2018-09-20:
Thanks for your advice, @adshar !

closed 2018-09-20 by BrianN92