microsoft/AutonomousDrivingCookbook #73: Black and white screen when running the PowerShell command

ivannson, 2018-10-25:
When I run the following in PowerShell:
`cd D:\AD_Cookbook_AirSim
 .\AD_Cookbook_Start_AirSim.ps1 landscape`

I get: 
`Configuring AirSim for scenario landscape...
Creating configuration JSON for scenario landscape...
Attempting to write configuration json to C:\Users\IvanNovikov\Documents\AirSim\set
Configuration json successfully written.
Starting AirSim for scenario landscape...
Executing D:\AD_Cookbook_AirSim\Landscape_Neighborhood\JustAssets.exe landscape...`

Then 1/2 black 1/2 white screen appears as shown in the picture and nothing happens
[image]

mitchellspryn, 2018-10-29:
Odd. This seems to be different from the linked issue, as I haven't seen this before. This is probably an issue with your graphics card drivers. Can you check the following:

1) Can you open a command prompt in the Landscape_Neighborhood and run "JustAssets.exe neighborhood?"

2) Can you run any of the official [airsim binaries](https://github.com/Microsoft/AirSim/releases)?

3) Can you verify that you are running the latest drivers for your graphics card?

ivannson, 2018-10-30:
@mitchellspryn Thanks for your response!

1. Opening the JustAssets.exe gives me exactly the same thing as described in the post

2. Yes, I have been able to run SimpleMaze

3. I have updated the drivers for Intel(R) Iris Graphics 6100, which didn't seem to change much.

I am trying to run this on MacBook Pro through Bootcamp

mitchellspryn, 2018-10-31:
I think there are two problems:

1) Our binaries generally depend on an external NVIDIA GPU being available. I doubt the integrated graphics chips would be able to handle the environments.
2) We haven't tested on non-windows platforms. I'd be suprised if they worked. 

If you can't get your hands on a windows box, I'd recommend spinning up an Azure NV VM.

samby707, 2018-12-21:
I am running the command 
.\AD_Cookbook_Start_AirSim.ps1 landscape
This is the output I get:
Configuring AirSim for scenario landscape...
Creating configuration JSON for scenario landscape...
Attempting to write configuration json to C:\Users\HarmanX\Documents\AirSim\settings.json...
Configuration json successfully written.
Starting AirSim for scenario landscape...
Executing D:\Sambuddha\AD_Cookbook_AirSim\Landscape_Neighborhood\JustAssets.exe landscape...
Fatal error!

[image]
 This same error is coming when I'm running the JustAssets file.