microsoft/AutonomousDrivingCookbook #76: TypeError: bad operand type for unary -: 'str'

danfouer, 2018-12-11:
Let's look at a sample batch. The steering angle is represented by the red line in the image:

In[4]:

def draw_image_with_label(img, label, prediction=None):
    theta = label * 0.69 #Steering range for the car is +- 40 degrees -> 0.69 radians
    line_length = 50
    line_thickness = 3
    label_line_color = (255, 0, 0)
    prediction_line_color = (0, 0, 255)
    pil_image = image.array_to_img(img, K.image_data_format(), scale=True)
    print('Actual Steering Angle = {0}'.format(label))
    draw_image = pil_image.copy()
    image_draw = ImageDraw.Draw(draw_image)
    first_point = (int(img.shape[1]/2),img.shape[0])
    second_point = (int((img.shape[1]/2) + (line_length * math.sin(theta))), int(img.shape[0] - (line_length * math.cos(theta))))
    image_draw.line([first_point, second_point], fill=label_line_color, width=line_thickness)
    
    if (prediction is not None):
        print('Predicted Steering Angle = {0}'.format(prediction))
        print('L1 Error: {0}'.format(abs(prediction-label)))
        theta = prediction * 0.69
        second_point = (int((img.shape[1]/2) + (line_length * math.sin(theta))), int(img.shape[0] - (line_length * math.cos(theta))))
        image_draw.line([first_point, second_point], fill=prediction_line_color, width=line_thickness)
    
    del image_draw
    plt.imshow(draw_image)
    plt.show()

[sample_batch_train_data, sample_batch_test_data] = next(train_generator)
for i in range(0, 3, 1):
    draw_image_with_label(sample_batch_train_data[0][i], sample_batch_test_data[i])

###################################################################
Then I got:

TypeError                                 Traceback (most recent call last)
<ipython-input-4-6955bc10ff72> in <module>()
     24     plt.show()
     25 
---> 26 [sample_batch_train_data, sample_batch_test_data] = next(train_generator)
     27 for i in range(0, 3, 1):
     28     draw_image_with_label(sample_batch_train_data[0][i], sample_batch_test_data[i])

~\Anaconda3\envs\py35\lib\site-packages\keras_preprocessing\image.py in __next__(self, *args, **kwargs)
   1329 
   1330     def __next__(self, *args, **kwargs):
-> 1331         return self.next(*args, **kwargs)
   1332 
   1333     def _get_batches_of_transformed_samples(self, index_array):

D:\AIProject\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in next(self)
    240         # so it can be done in parallel
    241 
--> 242         return self.__get_indexes(index_array)
    243 
    244     def __get_indexes(self, index_array):

D:\AIProject\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in __get_indexes(self, index_array)
    262                 x_images = x_images[self.roi[0]:self.roi[1], self.roi[2]:self.roi[3], :]
    263 
--> 264             transformed = self.image_data_generator.random_transform_with_states(x_images.astype(K.floatx()))
    265             x_images = transformed[0]
    266             is_horiz_flipped.append(transformed[1])

D:\AIProject\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in random_transform_with_states(self, x, seed)
    138             x = image.random_channel_shift(x,
    139                                      self.channel_shift_range,
--> 140                                      img_channel_axis)
    141         if self.horizontal_flip:
    142             if np.random.random() < 0.5:

~\Anaconda3\envs\py35\lib\site-packages\keras_preprocessing\image.py in random_channel_shift(x, intensity_range, channel_axis)
    195         Numpy image tensor.
    196     """
--> 197     intensity = np.random.uniform(-intensity_range, intensity_range)
    198     return apply_channel_shift(x, intensity, channel_axis=channel_axis)
    199 

TypeError: bad operand type for unary -: 'str'

############################################################

How should I do?

danfouer, 2018-12-11:
> Let's look at a sample batch. The steering angle is represented by the red line in the image:
> 
> In[4]:
> 
> def draw_image_with_label(img, label, prediction=None):
> theta = label * 0.69 #Steering range for the car is +- 40 degrees -> 0.69 radians
> line_length = 50
> line_thickness = 3
> label_line_color = (255, 0, 0)
> prediction_line_color = (0, 0, 255)
> pil_image = image.array_to_img(img, K.image_data_format(), scale=True)
> print('Actual Steering Angle = {0}'.format(label))
> draw_image = pil_image.copy()
> image_draw = ImageDraw.Draw(draw_image)
> first_point = (int(img.shape[1]/2),img.shape[0])
> second_point = (int((img.shape[1]/2) + (line_length * math.sin(theta))), int(img.shape[0] - (line_length * math.cos(theta))))
> image_draw.line([first_point, second_point], fill=label_line_color, width=line_thickness)
> 
> ```
> if (prediction is not None):
>     print('Predicted Steering Angle = {0}'.format(prediction))
>     print('L1 Error: {0}'.format(abs(prediction-label)))
>     theta = prediction * 0.69
>     second_point = (int((img.shape[1]/2) + (line_length * math.sin(theta))), int(img.shape[0] - (line_length * math.cos(theta))))
>     image_draw.line([first_point, second_point], fill=prediction_line_color, width=line_thickness)
> 
> del image_draw
> plt.imshow(draw_image)
> plt.show()
> ```
> [sample_batch_train_data, sample_batch_test_data] = next(train_generator)
> for i in range(0, 3, 1):
> draw_image_with_label(sample_batch_train_data[0][i], sample_batch_test_data[i])
> 
> ###################################################################
> Then I got:
> 
> TypeError Traceback (most recent call last)
> in ()
> 24 plt.show()
> 25
> ---> 26 [sample_batch_train_data, sample_batch_test_data] = next(train_generator)
> 27 for i in range(0, 3, 1):
> 28 draw_image_with_label(sample_batch_train_data[0][i], sample_batch_test_data[i])
> 
> ~\Anaconda3\envs\py35\lib\site-packages\keras_preprocessing\image.py in **next**(self, *args, **kwargs)
> 1329
> 1330 def **next**(self, *args, **kwargs):
> -> 1331 return self.next(*args, **kwargs)
> 1332
> 1333 def _get_batches_of_transformed_samples(self, index_array):
> 
> D:\AIProject\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in next(self)
> 240 # so it can be done in parallel
> 241
> --> 242 return self.__get_indexes(index_array)
> 243
> 244 def __get_indexes(self, index_array):
> 
> D:\AIProject\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in __get_indexes(self, index_array)
> 262 x_images = x_images[self.roi[0]:self.roi[1], self.roi[2]:self.roi[3], :]
> 263
> --> 264 transformed = self.image_data_generator.random_transform_with_states(x_images.astype(K.floatx()))
> 265 x_images = transformed[0]
> 266 is_horiz_flipped.append(transformed[1])
> 
> D:\AIProject\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in random_transform_with_states(self, x, seed)
> 138 x = image.random_channel_shift(x,
> 139 self.channel_shift_range,
> --> 140 img_channel_axis)
> 141 if self.horizontal_flip:
> 142 if np.random.random() < 0.5:
> 
> ~\Anaconda3\envs\py35\lib\site-packages\keras_preprocessing\image.py in random_channel_shift(x, intensity_range, channel_axis)
> 195 Numpy image tensor.
> 196 """
> --> 197 intensity = np.random.uniform(-intensity_range, intensity_range)
> 198 return apply_channel_shift(x, intensity, channel_axis=channel_axis)
> 199
> 
> TypeError: bad operand type for unary -: 'str'
> 
> ############################################################
> 
> How should I do?

closed 2018-12-11 by danfouer

Vege1994, 2019-01-03:
So do I, What should I do?Have you solve the problem?

Vege1994, 2019-01-03:
I reinstall keras==2.1.2, and I solve the problem now.