microsoft/AutonomousDrivingCookbook #80: TransportError: Retry connection over the limit

angel272shaw, 2019-01-01:
**Your issue may already be reported! Please make sure to search all open and closed issues before starting a new one.**

Please fill out the sections below so we can understand your issue better and resolve it quickly.

## Problem description
*(Please provide a 2-3 sentence description of your problem. Be concise to ensure this description is useful for future users who might run into the same issue.)*
I have trained the Model. When I run the TestModel.ipynb, it throws a error "TransportError: Retry connection over the limit".
Something I have to illustrate, I don't know how to make a connection with AirSim by PowerShell. When I cd to the directory which has the file AD_Cookbook_Start_AirSim.ps1( E:\airsimFiles\AD_Cookbook_AirSim), and run the command 
>.\AD_Cookbook_Start_AirSim.ps1 landscape -windowed
It just opened the file AD_Cookbook_Start_AirSim.ps1 without any other imformations.
And when I did these operations, meanwhile, I launched the UE4 with LandscapeMountains Project as the fellow picture shows:
[image]

ps: I have deployed the environment. I don't know how to make a connection between the jupyter and AirSim. It means I don't know how to control the vehicle by python code. And I have found the tutorials about it. If you have the experience, please tell me. Thanks a lot.

## Problem details
*(Please describe your problem in as much detail as possible here. Make sure to include screenshots, code snippets, error messages, links and anything else you think will help us understand your problem better. If applicable, please also provide us a list of steps to reproduce your problem.)*
TransportError                            Traceback (most recent call last)
<ipython-input-5-380ab5c5fa3a> in <module>()
----> 1 client.confirmConnection()
      2 client.enableApiControl(True)
      3 car_controls = CarControls()
      4 print('Connection established!')

D:\myPythonDemo\airsim_demo\AirSimClient.py in confirmConnection(self)
    158     def confirmConnection(self):
    159         print('Waiting for connection: ', end='')
--> 160         home = self.getHomeGeoPoint()
    161         while ((home.latitude == 0 and home.longitude == 0 and home.altitude == 0) or
    162                 math.isnan(home.latitude) or  math.isnan(home.longitude) or  math.isnan(home.altitude)):

D:\myPythonDemo\airsim_demo\AirSimClient.py in getHomeGeoPoint(self)
    167 
    168     def getHomeGeoPoint(self):
--> 169         return GeoPoint.from_msgpack(self.client.call('getHomeGeoPoint'))
    170 
    171     # basic flight control

C:\ProgramData\Anaconda3\lib\site-packages\msgpackrpc\session.py in call(self, method, *args)
     39 
     40     def call(self, method, *args):
---> 41         return self.send_request(method, args).get()
     42 
     43     def call_async(self, method, *args):

C:\ProgramData\Anaconda3\lib\site-packages\msgpackrpc\future.py in get(self)
     41                 else:
     42                     if isinstance(self._error, error.RPCError):
---> 43                         raise self._error
     44                     else:
     45                         raise error.RPCError(self._error)

TransportError: Retry connection over the limit

## Experiment/Environment details
* Tutorial used: *(For example, AirSimE2EDeepLearning, DistributedRL etc.)*
* Environment used: *(For example, landscape, city, hawaii etc.)*
* Versions of artifacts used (if applicable): *(For example, Python 3.5, Keras 2.1.2 etc.)*
Tutorial used: I run the TrainModel of AirSimE2EDeepLearning and trained the network successfully.
Environment: landscape
Versions of python: 3.6
Keras: 2.1.2
os: windows

LeroyLK, 2019-01-09:
> .\AD_Cookbook_Start_AirSim.ps1 landscape -windowed
> It just opened the file AD_Cookbook_Start_AirSim.ps1 without any other imformations.
> And when I did these operations, meanwhile, I launched the UE4 with LandscapeMountains Project as the fellow picture shows:
Do you mean that when you run the command " .\AD_Cookbook_Start_AirSim.ps1 landscape -windowed", then the UE4 with LandscapeMountains Project run ?
However when I run, a problem happenes like this:
[image]

**Then, I mutually run the "JustAssets.exe" in the folder "Landscape_Neighborhood"** then picture like this, looks lack something:
[image]

angel272shaw, 2019-01-09:
> > .\AD_Cookbook_Start_AirSim.ps1 landscape -windowed
> > It just opened the file AD_Cookbook_Start_AirSim.ps1 without any other imformations.
> > And when I did these operations, meanwhile, I launched the UE4 with LandscapeMountains Project as the fellow picture shows:
> > Do you mean that when you run the command " .\AD_Cookbook_Start_AirSim.ps1 landscape -windowed", then the UE4 with LandscapeMountains Project run ?
> > However when I run, a problem happenes like this:
> > [image]
> 
> **Then, I mutually run the "JustAssets.exe" in the folder "Landscape_Neighborhood"** then picture like this, looks lack something:
> [image]
> 
> BUT, when I run the TestModel.ipynb, it can run normally!
> 
> Then when I run the environment Windows landscapemountains.zip from https://github.com/Microsoft/AirSim/releases, the same problem happens as you (TransportError: Retry connection over the limit)
> 
> PS: my settings.json is:
> {
> "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
> "SettingsVersion": 1.0,
> "SimMode": "Car"
> 
> }

Remark:
When you run the command ".\AD_Cookbook_Start_AirSim.ps1 landscape -windowed", you need to run it on the cmd of VS2017, instead of the cmd of win10. And in my system, when i run the command, it just open the file "AD_Cookbook_Start_AirSim.ps1" with notebook.
And I solved the issue by opening the "*.exe" of the AirSim, then connect it by the  python code 
`__car_client.confirmConnection()` 
And there is another issue happened when i connected to AirSim. When I use the library of the github repository of "Microsoft/AutonomousDrivingCookbook" as fellow:
`from AirSimClient import *
......
client = CarClient()
client.confirmConnection()
`
then I failed to connect to AirSim. But when i succeeded when i use the library "airsim" of the github repository of "Microsoft/AirSim" as fellow:
`import airsim
client = sirsim.CarClient()
client.confirmConnection()
`

When i run the "JustAssets.exe" in the folder "Landscape_Neighborhood", I got the same scenario like you.

mitchellspryn, 2019-01-30:
Angel272Shaw, something looks wrong with the binaries you are running. The top bar should say "JustAssets", not "LandscapeMountains." Try redownloading.