microsoft/AutonomousDrivingCookbook #84: TypeError: bad operand type for unary -: 'str'

GetDarren, 2019-01-17:
This issue maybe silly,  I'm new to this area, Sorry to have your time.

When I run into the TrainModel.ipynb

```
def draw_image_with_label(img, label, prediction=None):
    theta = label * 0.69 #Steering range for the car is +- 40 degrees -> 0.69 radians
    line_length = 50
    line_thickness = 3
    label_line_color = (255, 0, 0)
    prediction_line_color = (0, 0, 255)
    pil_image = image.array_to_img(img, K.image_data_format(), scale=True)
    print('Actual Steering Angle = {0}'.format(label))
    draw_image = pil_image.copy()
    image_draw = ImageDraw.Draw(draw_image)
    first_point = (int(img.shape[1]/2),img.shape[0])
    second_point = (int((img.shape[1]/2) + (line_length * math.sin(theta))), int(img.shape[0] - (line_length * math.cos(theta))))
    image_draw.line([first_point, second_point], fill=label_line_color, width=line_thickness)
    
    if (prediction is not None):
        print('Predicted Steering Angle = {0}'.format(prediction))
        print('L1 Error: {0}'.format(abs(prediction-label)))
        theta = prediction * 0.69
        second_point = (int((img.shape[1]/2) + (line_length * math.sin(theta))), int(img.shape[0] - (line_length * math.cos(theta))))
        image_draw.line([first_point, second_point], fill=prediction_line_color, width=line_thickness)
    
    del image_draw
    plt.imshow(draw_image)
    plt.show()

[sample_batch_train_data, sample_batch_test_data] = next(train_generator)
for i in range(0, 3, 1):
    draw_image_with_label(sample_batch_train_data[0][i], sample_batch_test_data[i])
```

The error message as below:

---------------------------------------------------------------------------
```
TypeError                                 Traceback (most recent call last)
<ipython-input-13-6955bc10ff72> in <module>
     24     plt.show()
     25 
---> 26 [sample_batch_train_data, sample_batch_test_data] = next(train_generator)
     27 for i in range(0, 3, 1):
     28     draw_image_with_label(sample_batch_train_data[0][i], sample_batch_test_data[i])

~\AppData\Local\conda\conda\envs\my_airsim\lib\site-packages\keras_preprocessing\image.py in __next__(self, *args, **kwargs)
   1524 
   1525     def __next__(self, *args, **kwargs):
-> 1526         return self.next(*args, **kwargs)
   1527 
   1528     def _get_batches_of_transformed_samples(self, index_array):

D:\AirSim\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in next(self)
    240         # so it can be done in parallel
    241 
--> 242         return self.__get_indexes(index_array)
    243 
    244     def __get_indexes(self, index_array):

D:\AirSim\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in __get_indexes(self, index_array)
    262                 x_images = x_images[self.roi[0]:self.roi[1], self.roi[2]:self.roi[3], :]
    263 
--> 264             transformed = self.image_data_generator.random_transform_with_states(x_images.astype(K.floatx()))
    265             x_images = transformed[0]
    266             is_horiz_flipped.append(transformed[1])

D:\AirSim\AutonomousDrivingCookbook\AirSimE2EDeepLearning\Generator.py in random_transform_with_states(self, x, seed)
    138             x = image.random_channel_shift(x,
    139                                            self.channel_shift_range,
--> 140                                      img_channel_axis)
    141         if self.horizontal_flip:
    142             if np.random.random() < 0.5:

~\AppData\Local\conda\conda\envs\my_airsim\lib\site-packages\keras_preprocessing\image.py in random_channel_shift(x, intensity_range, channel_axis)
    199         Numpy image tensor.
    200     """
--> 201     intensity = np.random.uniform(-intensity_range, intensity_range)
    202     return apply_channel_shift(x, intensity, channel_axis=channel_axis)
    203 

TypeError: bad operand type for unary -: 'str'
```

It seems the data type of "intensity_range" is 'str', which should be a number.
but I cannot find where the **channel_shift_range** been assigned as 'str'
Please help me out.

ivannson, 2019-01-18:
What version of Keras are you using? As suggested in #76, try reinstalling Keras and specify version 2.1.2 by running `pip install keras==2.1.2`

GetDarren, 2019-01-18:
> What version of Keras are you using? As suggested in #76, try reinstalling Keras and specify version 2.1.2 by running `pip install keras==2.1.2`

Thanks for your help,  I was using keras==2.2.4. 
After I change the version of  keras 2.1.2,  it works well.

mitchellspryn, 2019-01-30:
[This issue is relevant](https://github.com/Microsoft/AutonomousDrivingCookbook/issues/1).

closed 2019-01-30 by mitchellspryn

AndersonBY, 2019-03-29:
The new version of keras.preprocessing.image.ImageDataGenerator has a different order of params.
It add a **brightness_range** between **height_shift_range** and **shear_range**

```
keras.preprocessing.image.ImageDataGenerator(featurewise_center=False,  
                                             samplewise_center=False, 
                                             featurewise_std_normalization=False, 
                                             samplewise_std_normalization=False, 
                                             zca_whitening=False, 
                                             zca_epsilon=1e-06, 
                                             rotation_range=0, 
                                             width_shift_range=0.0, 
                                             height_shift_range=0.0, 
                                             brightness_range=None, 
                                             shear_range=0.0, 
                                             zoom_range=0.0, 
                                             channel_shift_range=0.0, 
                                             fill_mode='nearest', 
                                             cval=0.0, 
                                             horizontal_flip=False, 
                                             vertical_flip=False, 
                                             rescale=None, 
                                             preprocessing_function=None, 
                                             data_format=None, 
                                             validation_split=0.0, 
                                             dtype=None)
```
 So you can rewrite the DriveDataGenerator in Generator.py file like this to solve it:

```
super(DriveDataGenerator, self).__init__(
                 featurewise_center=featurewise_center,
                 samplewise_center=samplewise_center,
                 featurewise_std_normalization=featurewise_std_normalization,
                 samplewise_std_normalization=samplewise_std_normalization,
                 zca_whitening=zca_whitening,
                 zca_epsilon=zca_epsilon,
                 rotation_range=rotation_range,
                 width_shift_range=width_shift_range,
                 height_shift_range=height_shift_range,
                 shear_range=shear_range,
                 zoom_range=zoom_range,
                 channel_shift_range=channel_shift_range,
                 fill_mode=fill_mode,
                 cval=cval,
                 horizontal_flip=horizontal_flip,
                 vertical_flip=vertical_flip,
                 rescale=rescale,
                 preprocessing_function=preprocessing_function,
                 data_format=data_format)

```

wonjoonSeol, 2019-04-09:
> The new version of keras.preprocessing.image.ImageDataGenerator has a different order of params.
> It add a **brightness_range** between **height_shift_range** and **shear_range**
>
> ``` [...]

This is great, thank you!