microsoft/AutonomousDrivingCookbook #85: TestModel notebook stuck after connecting with airsim

wonjoonSeol, 2019-01-21:
**Your issue may already be reported! Please make sure to search all open and closed issues before starting a new one.**

Please fill out the sections below so we can understand your issue better and resolve it quickly.

## Problem description
After running second box - despite printing out "connection successful" 
the next box does not run. The following boxes simply have * mark.

## Problem details
[image]
Next lines do not run, stuck.

Running separate window of airsim using the following command: 
`powershell.exe -ExecutionPolicy Bypass -File .\AD_Cookbook_Start_AirSim.ps1 landscape -windowed
`
## Experiment/Environment details
* Tutorial used: AirSimE2EDeepLearning
* Environment used: landscape
* Versions of artifacts used (if applicable): Python 3.6.8, Keras 2.1.2, CuDDN 7.4

wonjoonSeol, 2019-01-30:
Just to let you know python script that overrides Carclient call in Share.scripts_downpour.app.airsim_client is the cause. 

Running script is fine if using the latest airsim client API instead of this script.
Note to other people experiencing issue with the notebooks. Scripts here needs an update in general (Including the keras version it supports) so don't use the codes here as it is.

mitchellspryn, 2019-01-30:
I tried this on two separate machines, and was unable to reproduce. Are you using the binaries linked in the tutorial, or the ones from the AirSim repo? They are not the same.

New2Coding2019, 2019-01-30:
@mitchellspryn This tutorial is really awesome! Much better than the documentation from CARLA. I have the same issue with the notebook. I first open a powershell and start the simulator (work and car appears) then I run the jupyter notebook for testing and it tells me the connection could be established but the notebooks gets stuck after running the third cell. I had no problems with the first and second notebook after downgrading keras to 2.1.2 (It would be nice to add this information to the main jupyter notebook https://github.com/Microsoft/AutonomousDrivingCookbook/tree/master/AirSimE2EDeepLearning with Install Keras).

Is it necessary to `pip install airsim` as proposed in the main documentation?

# Environment
- Windows 10, Python 3.4.5, keras=2.1.2, tensorflow-gpu on Nvidia Quadro P2000
- Downloaded the binaries linked in the tutorial (https://airsimtutorialdataset.blob.core.windows.net/e2edl/AD_Cookbook_AirSim.7z) extracted it on my desktop
- Downloaded the AirSim-master repo from github for the PythonClient stuff and replaced `../.../PythonClient/` by the path to the extracted repo file
- I did not use Anaconda but pure Python 3.4.5 with a virtual environment with all the necessary installations

New2Coding2019, 2019-01-30:
@wonjoonSeol Could you post your notebook which was executable?

mitchellspryn, 2019-01-30:
I didn't realize airsim is now pip installable. Don't install it.

Please use the AirSimClient.py in the repo. Don't use the one in the AirSim master repo.

Here's why: AirSim is rapidly changing. As you've noticed, many of the changes are not backwards compatible. Our goal here is to create a stable build that works everywhere. Hence, we have created a binary from a snapshot of the AirSim repo, and took a snapshot of their client libraries. You should have everything you need to run inside this repo.

New2Coding2019, 2019-01-30:
@mitchellspryn Thank you a lot for your support! But what about his part? 
```shell
if ('../../PythonClient/' not in sys.path):
    sys.path.insert(0, '../../PythonClient/')
```
Do I have to change something in these lines?

BTW I am very greatful to you guys for sharing AirSim and your experience with us! The tutorial is the best tutorial about self driving cars that I have seen. Only the part with the data cooking seems to be magic for programming beginners like myself :D.

mitchellspryn, 2019-01-30:
I'm not sure why that is there. It shouldn't do anything. Does it break something?

New2Coding2019, 2019-01-31:
@mitchellspryn Hmm ... I only changed the path to the model. And the remaining code is copied from your notebook. I then go into the folder of the `AD_Cookbook_AirSim` I open the powershell in this folder and execute `.\AD_Cookbook_Start_AirSim.ps1 landscape -windowed`. The simulator opens up with the car standing still (I can still control it with the keyboard). Then I run cell 1 and cell 2 of the notebook and everything works without errors the second cell generates this output

```shell
WARNING:tensorflow:From c:\pyenvs\airsim\lib\site-packages\keras\backend\tensorflow_backend.py:1264: calling reduce_prod (from tensorflow.python.ops.math_ops) with keep_dims is deprecated and will be removed in a future version.
Instructions for updating:
keep_dims is deprecated, use keepdims instead
WARNING:tensorflow:From c:\pyenvs\airsim\lib\site-packages\keras\backend\tensorflow_backend.py:1349: calling reduce_mean (from tensorflow.python.ops.math_ops) with keep_dims is deprecated and will be removed in a future version.
Instructions for updating:
keep_dims is deprecated, use keepdims instead
Waiting for connection: 
Connection established!
```

But when running cell 3 the notebook gets stuck.

Gurtaj, 2019-02-07:
@New2Coding2019 Did this issue resolve for you ? I am encountering the same problem as you. In my case, I was using the repo for this project and not the master but still having issues cell 3. As @wonjoonSeol  had mentioned that some files in  the Deep Reinforcement learning folder were causing the hang, I tried removing that folder from the copy of the repo I have on my computer. After that cell 2 started having the same problem (getting stuck and never executing).

wonjoonSeol, 2019-03-15:
Apologies for replying late, I have been working on this on a leisurely pace.

1.
I have managed updated the whole distributed codebase for the latest Airsim (v1.2.1). This version is superior in my opinion because it has less bugs and have supports for clockspeed. Which I think is a pretty big deal for RL learning.
 
I wasn't able to update AirSimE2EDeepLearning because the latest landscape map does not have any snows there. We cannot use same the training images in the newer version.

In terms of sharing the codes, I may not be able to share for a few months because I am currently working on this for a university project. I will ask my supervisor whether I could share my personal project on a public repository before the deadline. But if you need to update the current codes for the newest version before uploading my updates, I can give you some guidance.

You will need to update rewards.txt and roadlines.txt because the change in the map base coordinates. I have worked out the offset. [80.45703888, -122.9101944]. Apply this offset for v1.2.1 version.

2.
Those who have problems with Jupyter getting stuck, I suggest you copy the codes to a separate python file. Either 1) most of your problems will disappear 2) Give you proper error messages.

The codes work fine for both AirSimE2EDeepLearning and Distributed RL Local training with the exception of test_training for AirSimE2EDeepLearning.

That one alone gives "cuDNN failed to initialize" error.  Apparently, this error is raised when there is a compatibility issue with tensorflow-gpu and cudnn and cuda versions. However I don't get the same error when I run distributed RL.

@mitchellspryn, You have given us the version for Keras 2.1.2 from another issue.
Can you please also give us your tensorflow-gpu, cudnn and cuda version for running AirSimE2EDeepLearning.

In case that also doesn't work, if you kindly upload Youtube video on runing the test_model results would be very helpful indeed.

For the provided codebase, I am currently using:
- Tensorflow 1.13
- Keras 2.1.2 (But in my updated code I am using Keras 2.2.4 now)
- CUDA 10

tfromb, 2019-03-28:
@wonjoonSeol @Gurtaj @mitchellspryn Did you manage to solve the issue with the system crashing after it says the connection is successful? I am using AirSimE2EDeepLearning.

wonjoonSeol, 2019-04-03:
What do you mean by system crashing. I am unable to connect to E2EDeepLearning's Airsim Client using RunModel notebook.

tfromb, 2019-04-03:
@wonjoonSeol I mean the same issue you experienced with AirSim being stuck after connecting it via the TestModel. Mine says 'connection established' then when running the cells after this, AirSim freezes.

wonjoonSeol, 2019-04-03:
@tfromb yes, if you copy all the code and run it on python script instead you will probably see same information as above.

rishchou, 2019-11-29:
Is there a fix for this issue?

danialvi, 2022-02-23:
> 
About this:
if ('c:/Users/username/Desktop/AirSim/AirSim-master/PythonClient/' not in sys.path):
    sys.path.insert(0, 'c:/Users/username/Desktop/AirSim/AirSim-master/PythonClient/')

which file is it, and where is it located.

danialvi, 2022-03-18:
> @wonjoonSeol I mean the same issue you experienced with AirSim being stuck after connecting it via the TestModel. Mine says 'connection established' then when running the cells after this, AirSim freezes.

Did you solve the issue?