microsoft/AutonomousDrivingCookbook #89: Error in train.ipynb

hrishikeshtawade04, 2019-02-12:
On running the Train model code in Autonomous Driving cookbook I am getting following error in the third code cell. It seems there is an error in the Generators.py code. Kindly help.

-------------------------------------------------------------------------
ValueError                              Traceback (most recent call last)
<ipython-input-5-9955e8d83922> in <module>()
----> 1 data_generator = DriveDataGenerator(rescale=1./255., horizontal_flip=True, brighten_range = (0.4,0.0))
      2 train_generator = data_generator.flow    (train_dataset['image'], train_dataset['previous_state'], train_dataset['label'], batch_size=batch_size, zero_drop_percentage=0.95, roi=[76,135,0,255])
      3 eval_generator = data_generator.flow    (eval_dataset['image'], eval_dataset['previous_state'], eval_dataset['label'], batch_size=batch_size, zero_drop_percentage=0.95, roi=[76,135,0,255])

~/enpm-809k/AutonomousDrivingCookbook/AirSimE2EDeepLearning/Generator.py in __init__(self, featurewise_center, samplewise_center, featurewise_std_normalization, samplewise_std_normalization, zca_whitening, zca_epsilon, rotation_range, width_shift_range, height_shift_range, shear_range, zoom_range, channel_shift_range, fill_mode, cval, horizontal_flip, vertical_flip, rescale, preprocessing_function, data_format, brighten_range)
     45                  rescale,
     46                  preprocessing_function,
---> 47                  data_format)
     48         self.brighten_range = brighten_range
     49 

~/anaconda3/envs/airsim/lib/python3.5/site-packages/keras/preprocessing/image.py in __init__(self, featurewise_center, samplewise_center, featurewise_std_normalization, samplewise_std_normalization, zca_whitening, zca_epsilon, rotation_range, width_shift_range, height_shift_range, brightness_range, shear_range, zoom_range, channel_shift_range, fill_mode, cval, horizontal_flip, vertical_flip, rescale, preprocessing_function, data_format, validation_split, dtype)
    464             data_format=data_format,
    465             validation_split=validation_split,
--> 466             **kwargs)
    467 
    468 

~/anaconda3/envs/airsim/lib/python3.5/site-packages/keras_preprocessing/image/image_data_generator.py in __init__(self, featurewise_center, samplewise_center, featurewise_std_normalization, samplewise_std_normalization, zca_whitening, zca_epsilon, rotation_range, width_shift_range, height_shift_range, brightness_range, shear_range, zoom_range, channel_shift_range, fill_mode, cval, horizontal_flip, vertical_flip, rescale, preprocessing_function, data_format, validation_split, interpolation_order, dtype)
    360                 raise ValueError(
    361                     '`brightness_range should be tuple or list of two floats. '
--> 362                     'Received: %s' % (brightness_range,))
    363         self.brightness_range = brightness_range
    364 

ValueError: `brightness_range should be tuple or list of two floats. Received: 0.0

Misslindalian, 2019-02-16:
Hello, I have encountered the same problem. Have you solved this problem?Thanks a lot！

hrishikeshtawade04, 2019-02-16:
I think there is a problem of keras version

Misslindalian, 2019-02-16:
So how to solve it? Just upgrade keras version？Thanks a lot！

hrishikeshtawade04, 2019-02-16:
Ya. That should work. Haven't tried it. But one of my colleague have done
it. Jist check the issue section of airsim

Misslindalian, 2019-02-16:
Airsim？Can you tell me in detail?

hrishikeshtawade04, 2019-02-16:
Haven't tried it though. I have directly used the trained models. But take
a look at all issues. The solution is in one of them

Misslindalian, 2019-02-18:
Hello! How about your trained models? My tested result is not good. When I trained the models, it stopped when the iterator = 45.  But in fact, it needs to run 500 iterators. What I mean is that it "stopped early". Thank you for your patience!

hrishikeshtawade04, 2019-02-18:
What error did it show when it stopped at 45?

mitchellspryn, 2019-02-18:
For the initial keras error, make sure you are using version 2.1.2. [See this issue as to why](https://github.com/Microsoft/AutonomousDrivingCookbook/issues/1).

It probably stopped early because of the [EarlyStopping Callback](https://keras.io/callbacks/#earlystopping). [Here is why you want to use that callback](https://en.wikipedia.org/wiki/Early_stopping). 

If it actually showed an error, open a new issue. Please use the issue template and fill out all relevant forms.

Misslindalian, 2019-02-19:
Thank you for your answer! However, the model does not perform well. It can only run normally in the first few seconds. When it turns, it will collide and rush into the grass.

mitchellspryn, 2019-02-19:
Try retraining with a different RNG seed. There isn't enough data to guarantee that the network fully converges every time. But after a few attempts, you should get one that works.

OrSh1234, 2019-04-09:
Can we go back please to the original post about Keras brightness_range ? 
I have the same problem (I moved the notebook to a simple py file and run it) :

C:\AirSim_RL>python  TrainModel.py
Using TensorFlow backend.
Traceback (most recent call last):
  File "TrainModel.py", line 44, in <module>
    data_generator = DriveDataGenerator(rescale=1./255., horizontal_flip=True, brighten_range=(0.0,0.0))#brighten_range=0.4)
  File "C:\AirSim_RL\Generator.py", line 93, in __init__
    data_format)
  File "C:\Python\Python35\lib\site-packages\keras\preprocessing\image.py", line 466, in __init__
    **kwargs)
  File "C:\Python\Python35\lib\site-packages\keras_preprocessing\image\image_data_generator.py", line 362, in __init__
    'Received: %s' % (brightness_range,))
ValueError: `brightness_range should be tuple or list of two floats. Received: 0.0

Any help will be appreciated...

hrishikeshtawade04, 2019-04-09:
@Orsh1234 just change the keras version to 2.1.2 using the command conda install keras==2.1.2 and it solves the problem

OrSh1234, 2019-04-30:
hrishikeshtawade04 - thanks. It solves the problem.