microsoft/AutonomousDrivingCookbook #90: Modifications for brake and throttle prediction

Andrenyoto22, 2019-02-22:
**Your issue may already be reported! Please make sure to search all open and closed issues 

## Problem description
*Modification for break and throttle prediction 

## Problem details
*i've been trying to do modifications on the code so the code will predict brake or throttle isntead of steering. The throttle prediction works but still have very bad result and for break the mapping data always null. 
I do analysis for the code but got many things that i'm not fully uderstand yet. I would be so helpful if you can give me some guidance. Thank you

## Experiment/Environment details
* Tutorial used: AirSimE2EDeepLearning
* Environment used: landscape
* Versions of artifacts used (if applicable): Python 3.5, Keras 2.1.2

mitchellspryn, 2019-03-29:
Can you clarify your question? I'm not sure what you are asking. How did you modify the code? What do you mean by "mapping data always null?"

Modifying the code to predict throttle as well as steering is not a trivial task. For example, consider that it is impossible to determine the proper throttle from a single image - you'll need to feed a series of images into the net. You also have to optimize for two variables, so your reward function will need to take that into account.

closed 2019-04-04 by mitchellspryn

aldaoctavia, 2020-08-13:
can i see u code?