microsoft/AutonomousDrivingCookbook #91: eval.h5 & test.h5 not generated in data_cook

tfromb, 2019-03-04:
Im doing the E2EDeepLearning tutorial. When I try to read in the datasets in TrainModel from the exploration phase, the eval.h5 and test.h5 files are not generated and so can not be opened. I've gone back to the DataExploration phase and run the code, however, they still aren't generating. How can I create them?

Thanks

`OSError                                   Traceback (most recent call last)
<ipython-input-10-2f2a51e01507> in <module>
      1 train_dataset = h5py.File(os.path.join(COOKED_DATA_DIR, 'train.h5'), 'r')
----> 2 eval_dataset = h5py.File(os.path.join(COOKED_DATA_DIR, 'eval.h5'), 'r')
      3 test_dataset = h5py.File(os.path.join(COOKED_DATA_DIR, 'test.h5'), 'r')
      4 
      5 num_train_examples = train_dataset['image'].shape[0]

~\Anaconda3\lib\site-packages\h5py\_hl\files.py in __init__(self, name, mode, driver, libver, userblock_size, swmr, **kwds)
    310                  track_order=None,
    311                  **kwds):
--> 312         """Create a new file object.
    313 
    314         See the h5py user guide for a detailed explanation of the options.

~\Anaconda3\lib\site-packages\h5py\_hl\files.py in make_fid(name, mode, userblock_size, fapl, fcpl, swmr)
    140         plist = h5p.create(h5p.FILE_CREATE)
    141         plist.set_link_creation_order(
--> 142             h5p.CRT_ORDER_TRACKED | h5p.CRT_ORDER_INDEXED)
    143         plist.set_attr_creation_order(
    144             h5p.CRT_ORDER_TRACKED | h5p.CRT_ORDER_INDEXED)

h5py\_objects.pyx in h5py._objects.with_phil.wrapper()

h5py\_objects.pyx in h5py._objects.with_phil.wrapper()

h5py\h5f.pyx in h5py.h5f.open()

OSError: Unable to open file (unable to open file: name = 'C:/Users/zs70/Documents/Tomi/Airsim/AutonomousDrivingCookbook-master/data_cooked/eval.h5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0)
`

wonjoonSeol, 2019-03-15:
It works fine for me, check if you are using Keras 2.1.2, you cannot use newer version of Keras on AirSimE2EDeepLearning. It should be written on the readme really. I will update..

mitchellspryn, 2019-03-29:
Are you on python 3.7? [If so, that's your issue](https://github.com/Microsoft/AutonomousDrivingCookbook/issues/94).

closed 2019-04-03 by tfromb