microsoft/AutonomousDrivingCookbook #98: AzCopy Transfer from file to directory path not supported - Help.

tfromb, 2019-03-27:
## Problem description
 I am using AzCopy to download the stand alone simulator environment. However, when I run the AzCopy code I get an error about transfer from file to directory not being supported. What destination is acceptable for AzCopy? I've read the AzCopy documentation however can't seem to find anything on this specifically. Thank you

## Problem details
This is what I inputted:
 C:\Program Files (x86)\Microsoft SDKs\Azure>AzCopy /Source:https://airsimtutorialdataset.blob.core.windows.net/e2edl/AD_Cookbook_AirSim.7z /Dest:C:/Users/zs70/Documents/Tomi/Airsim/AutonomousDrivingCookbook-master/Landscape

Below is the output I get back:

```
[2019/03/27 14:21:09][WARNING] The command line "AzCopy /Source:https://airsimtutorialdataset.blob.core.windows.net/e2edl/AD_Cookbook_AirSim.7z /Dest:C:/Users/zs70/Documents/Tomi/Airsim/AutonomousDrivingCookbook-master" in the journal file "C:\Users\zs70\AppData\Local\Microsoft\Azure\AzCopy\AzCopy.jnl" is different from your input.
[2019/03/27 14:21:09][WARNING] Incomplete operation with different command line detected at the journal directory "C:\Users\zs70\AppData\Local\Microsoft\Azure\AzCopy".
Do you want to overwrite the journal to start a new operation? Choose Yes to overwrite, choose No to cancel current operation. (Yes/No) y
[2019/03/27 14:30:36][ERROR] Error parsing destination location "C:/Users/zs70/Documents/Tomi/Airsim/AutonomousDrivingCookbook-master": Transfer from a file to a directory path is not supported. Please update /Dest with an absolute path to a file. For more details, please type "AzCopy /?:Dest" or use verbose option /V.
```

# Experiment/Environment details
* Tutorial used: AirSimE2EDeepLearning
* Environment used: Landscape
* Versions of artifacts used (if applicable): Python 3.7, Keras 2.1.2

closed 2019-03-28 by tfromb

DavidGakwaya, 2020-02-13:
Hi, did you manage to get the solution to this?

shaktiug, 2021-08-31:
This is the solution : -
The destination is interpreted according to following rules:
                              1) When source is a single file, destination
                              is interpreted as a single file.
                              2) When source is a directory, destination
                              is interpreted as a directory.