microsoft/AutonomousDrivingCookbook #102: ValueError: Received an empty batch. Batches should at least contain one item.

Spatialghost, 2019-04-10:
Hi, i started train model with custom dataset and after 24 epochs i've got error:
Traceback (most recent call last):
  File "train_model.py", line 94, in <module>
    validation_data=eval_generator, validation_steps=num_eval_examples//batch_size, verbose=2)
  File "C:\Python36\lib\site-packages\keras\legacy\interfaces.py", line 87, in wrapper
    return func(*args, **kwargs)
  File "C:\Python36\lib\site-packages\keras\engine\training.py", line 2169, in fit_generator
    use_multiprocessing=use_multiprocessing)
  File "C:\Python36\lib\site-packages\keras\legacy\interfaces.py", line 87, in wrapper
    return func(*args, **kwargs)
  File "C:\Python36\lib\site-packages\keras\engine\training.py", line 2305, in evaluate_generator
    raise ValueError('Received an empty batch. '
ValueError: Received an empty batch. Batches should at least contain one item.

batch_size=32

How to solve this problem?

mitchellspryn, 2019-04-10:
Is your batch size a multiple of 32?

Spatialghost, 2019-04-12:
> 
> 
> Is your batch size a multiple of 32?

I use code without any changes.

Spatialghost, 2019-04-12:
Solved!
Just have to restart.
Different problem now, early stopping after 90 - 100 epochs.

martsalz, 2019-11-08:
What did you have to restart? I have the problem on several machines.

kutilanak, 2022-05-13:
> Solved! Just have to restart. Different problem now, early stopping after 90 - 100 epochs.

AFAIK it doesn't have to complete the training to 500 epoch. If the val_loss did not improve after several epoch, it will automatically stop (early stopping)

u can use the latest model it saved.