microsoft/AutonomousDrivingCookbook #103: Steering data is collected as -1,0,1

CrookedNoob, 2019-04-11:
I am trying to collect the steering data on my own. I got ~1 GB of data and noticed that the steering values are -1,0,1. Sharing [airsim_rec.txt](https://github.com/Microsoft/AutonomousDrivingCookbook/files/3068322/airsim_rec.txt) for reference.

When I referred to the original AirSim data, the values are continuous. 

Is it because I am using keyboard to collect the data? 

Does this Sim support  any controller/joystick? 

Please let me know if there is any workaround

NextSim, 2019-04-16:
Yes.. the keyboard will only produce -1, 0 and 1. 

There are joysticks / controllers that are supported, you'd have to look at the documentation.  Alternatively, I bought a cheap joystick that wasn't supported, but I wrote a program that read in the joystick data and then used the airsim api to send that data to airsim.

closed 2019-04-30 by CrookedNoob

Misslindalian, 2019-06-04:
@NextSim Hello，Can you tell me what type of joystick you bought? Can you share the code to read the steering angle of the joystick? Thank you very much!