microsoft/AutonomousDrivingCookbook #106: TypeError: unsupported operand type(s) for *: 'ZMQIOLoop' and 'float'

gonzaloaguilarjimenez, 2019-05-02:
## Problem description
I am trying to run the TestModel and I have found a TypeError when connecting to AirSim. It seems to me that it could come from the Tornado version, but actually I have tried with more than one version and the problem continues.

## Problem details

Details of the problem are as follow: 

When I run the script .\AD_Cookbook_Start_Airsim.ps1 landscape, everything runs perfectly in UE4 and AirSim but once I run the Jupyter Notebook (Step 2: "Test the model") as follows, it is when I find the error after the second cell (posted below): 

model = load_model(MODEL_PATH)
client = CarClient()
client.confirmConnection()
client.enableApiControl(True)
car_controls = CarControls()
print('Connection established!')

Then I found the folowing error message: 

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-5-abfe90f3be35> in <module>
      1 model = load_model(MODEL_PATH)
      2 
----> 3 client = CarClient()
      4 client.confirmConnection()
      5 client.enableApiControl(True)

~\PycharmProjects\Cookbook\AutonomousDrivingCookbook-master\AutonomousDrivingCookbook-master\AirSimE2EDeepLearning\AirSimClient.py in __init__(self, ip)
    504         if (ip == ""):
    505             ip = "127.0.0.1"
--> 506         super(CarClient, self).__init__(ip, 42451)
    507 
    508     def setCarControls(self, controls):

~\PycharmProjects\Cookbook\AutonomousDrivingCookbook-master\AutonomousDrivingCookbook-master\AirSimE2EDeepLearning\AirSimClient.py in __init__(self, ip, port)
    148 class AirSimClientBase:
    149     def __init__(self, ip, port):
--> 150         self.client = msgpackrpc.Client(msgpackrpc.Address(ip, port), timeout = 3600)
    151 
    152     def ping(self):

~\Anaconda3\envs\Cookbook\lib\site-packages\msgpackrpc\client.py in __init__(self, address, timeout, loop, builder, reconnect_limit, pack_encoding, unpack_encoding)
     13 
     14         if timeout:
---> 15             loop.attach_periodic_callback(self.step_timeout, 1000) # each 1s
     16 
     17     @classmethod

~\Anaconda3\envs\Cookbook\lib\site-packages\msgpackrpc\loop.py in attach_periodic_callback(self, callback, callback_time)
     37 
     38         self._periodic_callback = ioloop.PeriodicCallback(callback, callback_time, self._ioloop)
---> 39         self._periodic_callback.start()
     40 
     41     def dettach_periodic_callback(self):

~\Anaconda3\envs\Cookbook\lib\site-packages\tornado\ioloop.py in start(self)
    885         self._running = True
    886         self._next_timeout = self.io_loop.time()
--> 887         self._schedule_next()
    888 
    889     def stop(self) -> None:

~\Anaconda3\envs\Cookbook\lib\site-packages\tornado\ioloop.py in _schedule_next(self)
    913     def _schedule_next(self) -> None:
    914         if self._running:
--> 915             self._update_next(self.io_loop.time())
    916             self._timeout = self.io_loop.add_timeout(self._next_timeout, self._run)
    917 

~\Anaconda3\envs\Cookbook\lib\site-packages\tornado\ioloop.py in _update_next(self, current_time)
    920         if self.jitter:
    921             # apply jitter fraction
--> 922             callback_time_sec *= 1 + (self.jitter * (random.random() - 0.5))
    923         if self._next_timeout <= current_time:
    924             # The period should be measured from the start of one call

TypeError: unsupported operand type(s) for *: 'ZMQIOLoop' and 'float'

## Experiment/Environment details
* Tutorial used: AirSimE2EDeepLearning
* Environment used:  landscape (all my environment are running perfectly)
* Versions of artifacts used: 
 -- Windows 10
 -- Anaconda
 -- Python 3.5
 -- Tornado 6.0.2

Thanks a lot in advance.

closed 2019-05-09 by gonzaloaguilarjimenez

scaperex, 2019-09-03:
@gonzaloaguilarjimenez may I ask how did you solve this issue?

gonzaloaguilarjimenez, 2019-09-03:
Good afternoon,

Yes, I solved it and it worked perfectly.

Thanks for the response.

Best regards.

scaperex, 2019-09-03:
@gonzaloaguilarjimenez would you mind sharing **how** you solved it?
Thanks!

scaperex, 2019-09-03:
downgrading torando worked for me. `conda install tornado=4.5.3`
https://github.com/jupyter/notebook/issues/3407#issue-303235000

Thanks anyways

gonzaloaguilarjimenez, 2019-09-03:
Hello, 

It was a problem with the anaconda version apparently. 

I just uninstalled and reinstalled Anaconda and then it worked. 

I do not remember the previous version, but if necessary I can check the current one.

Best regards.

christian33moreno, 2022-08-13:
I'm getting the same error. What version of anaconda ended up working?