microsoft/AutonomousDrivingCookbook #114: no  Cooking

higho51420, 2020-05-02:
**Your issue may already be reported! Please make sure to search all open and closed issues before starting a new one.**

Please fill out the sections below so we can understand your issue better and resolve it quickly.

## Problem description
*(Please provide a 2-3 sentence description of your problem. Be concise to ensure this description is useful for future users who might run into the same issue.)*
ImportError: No module named 'Cooking'
## Problem details
*(Please describe your problem in as much detail as possible here. Make sure to include screenshots, code snippets, error messages, links and anything else you think will help us understand your problem better. If applicable, please also provide us a list of steps to reproduce your problem.)*

## Experiment/Environment details
* Tutorial used: *(For example, AirSimE2EDeepLearning, DistributedRL etc.)*
* Environment used: *(For example, landscape, city, hawaii etc.)*
* Versions of artifacts used (if applicable): *(For example, Python 3.5, Keras 2.1.2 etc.)*

mitchellspryn, 2020-05-02:
Check your python path and make sure the cooking folder is on it.

higho51420, 2020-05-03:
@mitchellspryn 
thank you so much !

closed 2020-05-04 by higho51420