microsoft/AutonomousDrivingCookbook #118: Getting IndexError: list index out of range while running TrainModel.ipynb ([in 5]).

Eldart95, 2020-10-16:
## Problem description
Getting IndexError: list index out of range while running TrainModel.ipynb ([in 5]).

## Problem details
IndexError                                Traceback (most recent call last)
<ipython-input-20-7b982afee170> in <module>
      1 image_input_shape = sample_batch_train_data[0].shape[1:]
----> 2 state_input_shape = sample_batch_train_data[1].shape[1:]
      3 activation = 'relu'
      4 
      5 #Create the convolutional stacks

IndexError: list index out of range

## Experiment/Environment details
* Tutorial used: AirSimE2EDeepLearning/TrainModel.ipynb
* Environment used: none 
* Versions of artifacts used (if applicable): *(Python 3.6, Keras 2.1.2 ,TF 1.8)*

SriramKichugari, 2020-10-17:
I have the same issue. Have you find any solution for this?

Eldart95, 2020-10-17:
unfortunately not yet