microsoft/AutonomousDrivingCookbook #122: Kaggle AirSim End-to-End Learning to share

rkuo2000, 2021-11-28:
Just to share my port on Kaggle [AirSim End-to-End Learning](https://www.kaggle.com/rkuo2000/airsim-end-to-end-learning)

To make it work on Kaggle with Keras 2.6.0,
Cooking.py and Generator.py are modified in my [github fork](https://github.com/rkuo2000/AutonomousDrivingCookbook)

It seems I have better prediction than the tutorial notebook.

lus105, 2021-12-05:
Hi, I am using python 3.7.5, keras 2.6.0 and ft 2.6.0. However, while training I get the following error:
`Traceback (most recent call last):
  File "c:/Users/.../Desktop/virtual keras 2.6/Train_model.py", line 40, in <module>
    data_generator = DriveDataGenerator(rescale=1./255., horizontal_flip=True, brighten_range=0.4)
TypeError: __init__() got an unexpected keyword argument 'brighten_range'`
Any thoughts on solving this?

rkuo2000, 2021-12-06:
It seems you are running it on PC, because you have message of C:/Users/
My example is for Kaggle, just go to
https://kaggle.com/rkuo2000/airsim-end-to-end-learning   and do [Copy &
Edit] to replicate my project.